SmokeMonsterPacks/Mega-Sg-Jailbreak #1: Colecovision Choplifter bullet bug.

SegaSnatcher, 2019-04-02:
Bullets shoot out as dual lines instead of blocks.  

Video for example.   

[https://www.youtube.com/watch?v=2yVBCvj_xho](url)

frederic-mahe, 2019-04-02:
@SegaSnatcher could you please confirm that the correct ROM has one of the following hash values?
```
SHA256: bac1f22f8c210ca19d06994cb88f57988de25719fb818b12cc3a3541d9c85414
SHA1:   a90bcac7da24b518002971caec342ffdd43b2348
MD5:    782ddbf40366269591205506f73906a1
CRC32:  e7a1ff9f
```

closed 2019-04-02 by frederic-mahe

SegaSnatcher, 2019-04-02:
MD5:    B3719118E9CFCC1234CD01C1E3FB84CF

SHA1:   3A7E5EA50495A666BAC8DF0B3B2C82B5BE65AD18

CRC32:  030E0D48

frederic-mahe, 2019-04-03:
Hum, that means that the SMDB `Colecovision SMDB.txt` has a bad version of Choplifter. @SmokeMonsterPacks could you please check that? @SegaSnatcher what are the hash values for the good ROM?

SegaSnatcher, 2019-04-03:
Good Rom Hash Values

CRC32: 030E0D48
MD5:    B3719118E9CFCC1234CD01C1E3FB84CF
SHA-1:  3A7E5EA50495A666BAC8DF0B3B2C82B5BE65AD18

frederic-mahe, 2019-04-03:
Thanks @SegaSnatcher 

@SmokeMonsterPacks could you please check this? Maybe we could work on the ColecoVision pack and compare it to No-Intro's pack to find outliers like this one?