SmokeMonsterPacks/Mega-Sg-Jailbreak #2: Feature request: Master System 3D support

TheRealFlamepanther, 2019-04-02:
Probably a low priority, but it would be great to have support for the 3D SMS games. The original LCD glasses won't work with modern displays, but could maybe be driven by the headphone jack once the RGB adapter is released. But for modern displays, the HDMI standard has support for 3DTV. 

I believe 3DTV HDMI output is essentially streamed as side-by-side, and supports either 2x 1080p @ 24 fps or 2x 720p @ 60 fps. Obviously only the latter would be suitable here. Since SMS 3D originally works by alternating frames in synch with the glasses (60ish fps /2 for an effective 30 fps) the Sg would logically need to blank whichever half of the 720p side-by-side image is unused in each frame. This would also preserve the flickery nature of the original experience, which could be neat or obnoxious (or both) depending on one's point of view :)

frederic-mahe, 2019-04-02:
Interesting idea. That feature could apply to the official firmware too.

TheRealFlamepanther, 2019-04-02:
Yes, since certain JB fixes and features mysteriously find their way into the official version, this suggestion applies to both ;)

axebox, 2019-04-02:
VR glasses can also display 3D 1080p signals. Can you imagine plugging the SG into a VR setup?

SmokeMonsterPacks, 2019-04-02:
It's OK to report on the official fw here--there is no magic involved. Kev used the Super Nt tracker to diagnose and fix problems with the SNt OFW, so he's probably watching here too. 

I will close this request though since I'm sure the jailbreaker is aware of the SMS 3D and can choose on their own.

closed 2019-04-02 by SmokeMonsterPacks