SmokeMonsterPacks/Mega-Sg-Jailbreak #4: Sega CD Model 1 & Mickey Mania (USA) Graphical Glitches

SmokeMonsterPacks, 2019-04-02:
A few people have reported that playing Mickey Mania on a Sega CD Model 1 results in graphical glitches shortly into the first stage. I believe that this does not affect Model 2s, but could be mistaken.

The graphical crash is visible in this video at [53:42](https://youtu.be/_LVWm_3BYWA?t=3222)

## Steps to reproduce the problem
Jump on the moving crates in stage one.

## ROM name and hash value
Mickey_Mania_-_The_Timeless_Adventures___U_Darkwater.zip
SHA-1 65560108F0516212550C4C5203E2F6C11E603F09

## firmware version
JB7.3

frederic-mahe, 2019-04-03:
@SmokeMonsterPacks I cannot confirm it is a `valid ROM`, please provide the hash value of the unziped file.

frederic-mahe, 2019-04-04:
TOSEC gives the following hash values for the tracks of this Sega CD:
```
        <game name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)[!]">
                <description>Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)[!]</description>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 01 of 28)[!].iso" size="13649920" crc="5b2898ac" md5="38cdce0d298b661d40af152d6916af12" sha1="574a4d9bcaaf75b9728808abd7232d29f6e0aae8"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 02 of 28)[!].wav" size="22242908" crc="9395cf7d" md5="7c487170cf84e565b664a6caef54260a" sha1="dda2aadc193ca958a162188264af9d711edc0e23"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 03 of 28)[!].wav" size="22402844" crc="cde8bdf7" md5="14eb0365a3f5949a6437548f3f177e35" sha1="064ba5965528b163511ff88db0e80e88982bf0e5"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 04 of 28)[!].wav" size="21730172" crc="8e9385fa" md5="f0e62373a0fd5ee9cfed57f5cd1b2fbd" sha1="c2f5a56d8cad84a8c170521e164bba9dd0c022b1"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 05 of 28)[!].wav" size="22849724" crc="79136032" md5="0f3b088d32910247b77022402442cbf7" sha1="48111ef9baba5c5e51e51a9828aacdd46dae058e"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 06 of 28)[!].wav" size="21539660" crc="8c66d382" md5="59c94c810e7ca592f2b3808851739e2f" sha1="a5d276ffdd1070b5c6e7fac99646f5d26944148c"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 07 of 28)[!].wav" size="15490316" crc="d16c98a4" md5="4a7d17f71f939c16e28868936cb9ca34" sha1="9f82ca46818355c2cacc735f392e7dbf886ccec2"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 08 of 28)[!].wav" size="22946156" crc="044b9f2d" md5="6a4db08714a8fbcb18cadb44ee95cefd" sha1="f7b3dd245bad89ec5607ae599188fe2195f3bfd3"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 09 of 28)[!].wav" size="22885004" crc="89391554" md5="b4388f0dd61691c57b2ab280c1051128" sha1="3bacc305ee2d36194c638c77746f90190fb6daa9"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 10 of 28)[!].wav" size="22275836" crc="6a38349b" md5="f780eff2ef586d5f60f256339452d2c0" sha1="3508b8289d91986c4fee7ada6dd0e19082dd20a2"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 11 of 28)[!].wav" size="22786220" crc="f6d03962" md5="f58d58a3c9b4a91e9cfbf3fbe7c0cc1f" sha1="29c1bff068173881bf58144bc6bb586af068aecd"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 12 of 28)[!].wav" size="22946156" crc="b29075a8" md5="fe115293e8a9cd792069067f61b86b87" sha1="2217d0589d4a94ca41a0f1819287f1c4bd9751db"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 13 of 28)[!].wav" size="22595708" crc="313d754b" md5="c53556f6585eb2284738ad8fe1bc0476" sha1="2a16dab0794fc7716750c678cd37525d25d96df6"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 14 of 28)[!].wav" size="26467100" crc="35908462" md5="9eb3cd46fbae38e24a76117284b0082d" sha1="02810d0dd521edef7dba3e6792fe2f54cfdf4902"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 15 of 28)[!].wav" size="32483516" crc="ad63c92e" md5="c4f3ba12f1a2581d25de82e97c4e244c" sha1="44b1956cac078a054b66b8fb86e7030048e798ad"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 16 of 28)[!].wav" size="32194220" crc="41e39854" md5="257e5186dfbda0a6fcb71405cd368ee4" sha1="6b4d3a2508c97f8a0334b0785f9eb913fcd38f3d"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 17 of 28)[!].wav" size="22082972" crc="1b92d321" md5="13952ebcec1ab4c793aed44daee61cf3" sha1="2928fb000e22992aac52895aabffa65f170b4851"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 18 of 28)[!].wav" size="21890108" crc="77ff44c2" md5="a43770d20168cb7800aa70c25cec5ae6" sha1="75363cc408d3197aeb90743843faef7c83cea061"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 19 of 28)[!].wav" size="21890108" crc="44cdd61a" md5="42bfd45dc4cbec37c6d716a3d2fdf91f" sha1="0d2160a632893802e3002d87d261549700cb4714"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 20 of 28)[!].wav" size="22466348" crc="134b78ab" md5="0f65a998297673ccc614b3498c697c9f" sha1="e5f7913676ed8ae126003126dedd4adb1c1d7d52"/>
[... cut ...] 1a14"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)(Track 28 of 28)[!].wav" size="1764044" crc="97072686" md5="497eaa322763f874966674cc56061241" sha1="e077cfe6660cc809f1882865f9b5c33c6d271b57"/>
                <rom name="Mickey Mania - Timeless Adventures of Mickey Mouse (1994)(Sony Imagesoft)(NTSC)(US)[!].cue" size="5157" crc="4410b315" md5="8d5587efa60e93ca47746d0d7d0a3a7a" sha1="6c865d20f5244c65b2a9f1853d793eeb96f2c5f5"/>
```

zxtheproto, 2019-04-16:
According to Analogue's website, issues with the Sega CD port of Mickey Mania were fixed in firmware version 4.4.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-4

SmokeMonsterPacks, 2019-04-17:
This should be fixed in v7.4

closed 2019-04-17 by SmokeMonsterPacks