SmokeMonsterPacks/Mega-Sg-Jailbreak #5: Sega CD Model 1 & Snatcher (USA) Missing In-Game Text

SmokeMonsterPacks, 2019-04-02:
Snatcher is missing in-game text when used on my Sega CD Model 1. I don't have a real disc to test with, so it very well could be a problem with my burned media. Could someone please test with a real disc?

Missing text shown at 1:32:13 in [this video](https://youtu.be/8wXES-vBsCg?t=5533). 

## ROM name and hash value
Snatcher (1994)(Konami)(US) TOSEC.zip
SHA-1 D8DCC8F54BE4C4D01B2B88CF9A32B269B90DFF20

## firmware version
JB7.3

tjanas, 2019-04-02:
Curious, have you tried the [redump](https://archive.org/download/RedumpSegaMegaCdSegaCd20160731Part2/Snatcher%20%28USA%29.zip) copy of this game, or is the TOSEC image identical to it?

SmokeMonsterPacks, 2019-04-02:
Redump is BIN/CUE while TOSEC is CCD/IMG/SUB. Theoretically, TOSEC should be closer to the original disc since it contains the sub-channel data, but I could test Redump next week when I dig out my burning setup.

SmokeMonsterPacks, 2019-04-02:
Remowilliams confirmed this behavior with the real gold disc of Snatcher on his Sega CD Model 2. However, after cleaning the edge connector with deoxit the text returned. I did the same and now have text, so this is not a Mega Sg issue--just business as usual for Sega CDs :)

SmokeMonsterPacks, 2019-04-03:
OK, I've figured out that the text is only missing when using a region-free BIOS from the menu. The stock BIOS loads the text correctly.

wiggy2k, 2019-04-03:
nothing is stored in the subchannels for sega cd. . 
redump is bin + cue, tosec is iso + wav. this has nothing to do with the dump, rather to do with your bios implementation and the emulational layer.

wiggy2k, 2019-04-03:
oh actually the TOSEC dat entry for snatcher is actually img/ccd/sub but either way my comment still stands.

SmokeMonsterPacks, 2019-04-03:
Yeah, someone corrected me about that on CGD too.

GreatHierophant, 2019-04-03:
Sega CD games do not require the preservation of subchannel data, although it is the best practice to do so.  In the BIN/CUE format, the subchannel data must be designated within the cuesheet with the TRACK label MODE1/2352, not MODE1/2048.

SmokeMonsterPacks, 2019-04-17:
This should be fixed in v7.4

closed 2019-04-17 by SmokeMonsterPacks