SmokeMonsterPacks/Mega-Sg-Jailbreak #6: 6-button Mode not working for Controller 2 (Firmware 4.3/JB Firmware 7.3)

mrnarse, 2019-04-03:
On Official Firmware 4.3 and Jailbreak Firmware 7.3 the 6-button mode for Controller 2 is not working.

When doing the controller test in the Mega Sg menu, the X, Y, Z, and Mode buttons do not register for Controller 2. This has been tested with two 8Bitdo M30 2.4Ghz Controllers and a wired official 6-button controller.

The same behavior also appears when running a controller test ROM and was tested in 6-button games. (Tested on both the Mega Everdrive x7 and the Jailbreak Firmware Genesis core)

After flashing to Official Firmware 4.2 the issue is resolved, and 6-button mode works properly for Controller 2.

After flashing to Official Firmware 4.3 or Jailbreak Firmware 7.3 the issue reappears.

frederic-mahe, 2019-04-03:
Could someone please confirm that issue? If confirmed, I will tag it as a bug.

SegaSnatcher, 2019-04-03:
I tested and can confirm 2nd controller X,Y,Z buttons do not register with Jailbreak Firmware 7.3.

craigsblackie, 2019-04-03:
Yeah, can confirm on my unit too.

GreatHierophant, 2019-04-03:
For now, enabling Passthrough Mode will allow you to use the extra buttons on Controller 2.

frederic-mahe, 2019-04-16:
@mrnarse @SegaSnatcher @craigsblackie could you please try the new firmware and confirm that the issue is solved?

SmokeMonsterPacks, 2019-04-17:
This should be fixed in v7.4

closed 2019-04-17 by SmokeMonsterPacks