SmokeMonsterPacks/Mega-Sg-Jailbreak #7: Final Fight CD Random Character or Enemy lock up -Unverified-

chandlerdepMODE, 2019-04-04:
Final Fight CD Character or Enemy lock up, happens randomly but usually can get it to happen on the first or second level.  The redbook audio will continue to play and other enemies or players will continue to respond but I can't proceed.

I am using my original US game disc on a US Sega CD model 2 with original PSU's.  I can't reproduce the problem when I hook my Sega CD unit back up to my Model 1 VA6 Genesis.  I have tried both my original disc, plus a burned disc from the Redump set. 

Tried on both Official firmware 4.3 and Jailbreak 7.3 using both my Sega CD's BIOS and loading a region free bios'

Pictures show Damned (Thrasher) stuck on the floor and Cody mid kick

[image]
[image]
[image]
[image]

.
Cleaned the Genesis port with Deoxit on the Sega CD side
The only thing I have not done yet is recapped my system

frederic-mahe, 2019-04-04:
Thanks for this detailed bug report. Can someone please confirm that issue?

Vorenge, 2019-04-04:
This Glitch occurs on official 4.2 and 4.3 software even without the jailbreak on both the USA version and Japanese version. I tried to inform SmokeMonster during his live stream of the jailbreak when he was playing the game. The jailbreak is not the culprit of this bug.

frederic-mahe, 2019-04-04:
Thanks @Vorenge I marked the issue as a confirmed bug.

chandlerdepMODE, 2019-04-04:
Another detail that may or may not be helpful is, My Sega CD V2 is a later revision with a Sony Laser PCB, Several people on Discord could not produce the issue on their Model 2 units, it may be a possibility they have a different Rev model 2 CD unit?  Smokemonster seemed to not have the issue on his Model 1 unit

Vorenge, 2019-04-06:
I can confirm that the lockup occurs with both versions of the Sega CD model 2. I own two units, one made in Japan, and the other made in China. The lockup happens with the genuine version of the US release of Final Fight as well as a burned copy of the Japanese release. The lockup happens with the official Sega CD bios ver 2.0 on the Sega CD units, and the multi-region bios from the jailbreak firmware.
--
Japan Unit has:
Sega 315-5476A  3FD 6Z
Sega 316-5632 9329-S22
Toshiba RAM chips
--
China Unit has:
Sega 315-5476A  4ED 5D
Sega 315-5632  9321-216
Sharp RAM chips

chandlerdepMODE, 2019-04-16:
Upgraded to Jailbreak 7.4, I am no longer able to produce the issue, played until I ran out of continues without locking up.  Have not tried vanilla firmware 4.4 so can't confirm but it would make sense it was fixed. Thank you!

SmokeMonsterPacks, 2019-04-17:
Thanks for following up!

closed 2019-04-17 by SmokeMonsterPacks

Vorenge, 2019-04-17:
I will also confirm this has been corrected and the game now runs correctly on both of my model 2 SCD units.