SmokeMonsterPacks/Mega-Sg-Jailbreak #8: Unable to access in-game menu on Mega Everdrive (pre-X series model)

shawn, 2019-04-04:
## Expected behavior and actual behavior
I have an earlier model Mega Everdrive. Unsure if this would be considered the V1 or V2 model, but pre-X3/X5/X7 series. On a stock Genesis, pressing START+Down on the controller will open the in-game menu to access save state settings. On the Mega Sg, this key combination fails to switch into the in-game menu.

Here are two screenshots from the Mega Everdrive's settings, in case any options are relevant:

[image]

[image]

## Steps to reproduce the problem
Press START+Down during game play. The Mega Everdrive should switch into the in-game menu automatically.

## firmware version
Tested with both official firmware v4.2, 4.3 and Jailbreak Firmware v7.3.

Also tested using both an original wired 3-button Genesis controller and an 8bitdo 2.4g wireless controller, just to rule out it wasn't a specific controller issue.

## relevant Mega Sg settings
Also tested with both controllers in "Passthrough Mode" as well as "Hotkeys Disabled." Also ensured that no existing Mega Sg hotkey shortcuts were set to the same START+Down key combinations.

frederic-mahe, 2019-04-04:
Can someone please confirm this issue with earlier model Mega Everdrive?

GreatHierophant, 2019-04-04:
Yes, the in-game menu on the original Mega EverDrive (v1) does not work at this time.  The MegaEverDrive X7 was originally called the Mega EverDrive v2 and it's in-game menu works with a cartridge released under either name.

AC6y86, 2019-04-07:
I have a mega everdrive v2 and have the same issue, the menu doesn't come up.

GreatHierophant, 2019-04-07:
The Mega EverDrive v2/X7's in-game menu shares the default hotkey for bringing up the Mega Sg menu.  You must change the Mega Sg's menu hotkey to something other than the default of Down + Start. Also, some games will not work right with the Mega ED's in-game menu.

SmokeMonsterPacks, 2019-04-17:
Is this still present in v7.4?

shawn, 2019-04-17:
Retested on v7.4 and issue is still present. 

Attempted with both "Passthrough Mode," as well as "Hotkeys Disabled," settings toggled on and off.

Rustrock, 2019-06-29:
This issue is still present in v7.5.

farhanjiwani, 2019-12-19:
I have a Mega Everdrive v2 (pre-X series) running OS Version 2.01. I had to enable the in-game menu in options before it would work for me (as expected). Also, my Mega SG is on JB firmware 7.7.

On the MED screen with the files listed, press C and you will have access to `Options`. Just make sure `In-Game Menu` is set to `ON`

BamBam1963, 2023-03-28:
Same issue with my Mega Everdrive X5

tjanas, 2023-05-05:
@shawn @BamBam1963 @Rustrock @AC6y86 please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

closed 2023-06-25 by tjanas