SmokeMonsterPacks/Mega-Sg-Jailbreak #9: Feature request: SMS resolution option for GG core

WaferMouse, 2019-04-04:
Some emulators provide an option to run Game Gear games using the Master System resolution, it would be great to have this option on the SG. Sonic Triple Trouble is already very playable in this state, and it could be fun to hack other games to take advantage of it as well.

(I realise this is perhaps beyond the scope of the jailbreak, but maybe Analogue will notice.)

SegaSnatcher, 2019-04-05:
What's the benefit?   You can already scale GG to fill the screen vertically.

WaferMouse, 2019-04-05:
I guess a practical demonstration is best.

Sonic Triple Trouble:
https://www.youtube.com/watch?v=drcKKfWnYF8

Sonic Triple Trouble with Master System resolution:
https://www.youtube.com/watch?v=j33-qA9RGMI

frederic-mahe, 2019-04-05:
To summarize, it seems that using Master Sytem resolution "zooms out": a larger portion of the level is visible around Sonic.

TheRealFlamepanther, 2019-04-11:
This would be cool, but you wouldn't want it for every game. I notice the jailbreak uses special filename extensions to identify special cases. Maybe games that work at the expanded resolution could be similarly renamed?

GreatHierophant, 2019-04-13:
I do not believe this feature will be implemented.  It often shows glitches on portions of the screen the original hardware was never intended to display.  More importantly, the Mega Sg's scaler is built around a 160x144 resolution, which allows higher scaling (7.5x) for the graphics you are supposed to see and reduced borders.  The Game Gear as implemented in the Nt Mini uses Master System/NES scaling of a 256x240 resolution and does a maximum of 5x.

SmokeMonsterPacks, 2019-04-13:
Closing based on @GreatHierophant's comments.

closed 2019-04-13 by SmokeMonsterPacks