SmokeMonsterPacks/Mega-Sg-Jailbreak #10: SMB conversion has graphics problems

vanfanel, 2019-04-05:
HI there,

There's this SMB homebrew conversion to Genesis, available here:
http://sonicvaan.sonicgamesdimension.net/mirror/rare/MULS_1.4.2.BIN

However, it doesn't display the backgrounds on the MegaSG. Resetting the game makes it work sometimes, albeit with missing score and information.

Any ideas on what's going on?

frederic-mahe, 2019-04-06:
Hi @vanfanel does this homebrew game work on a Sega Genesis (through an Everdrive)?

vanfanel, 2019-04-06:
@frederic-mahe : Please ignore this report, it seems it doesn't work on real HW according to this thread on sega-16:
http://www.sega-16.com/forum/archive/index.php/t-12550.html
I will close this. Sorry for wasting your time.

vanfanel, 2019-04-06:
Reopened this one.
On the original development thread, there are reports that latest version (1.4.2, the one in the link) does work on real hw:
http://devster.proboards.com/thread/613?page=4

frederic-mahe, 2019-04-06:
I've just tested version 1.4.2 on my model 1 Genesis with an Everdrive X7 with the latest firmware. The game boots but only sprites are visible, like on a Mega Sg. I think the problem is coming from the game itself, and not from the FPGA reproduction of the console.

@vanfanel If the developers of that game make a new version that runs on original consoles but not on the Mega Sg, please feel free to re-open that issue.

closed 2019-04-06 by frederic-mahe

vanfanel, 2019-04-06:
@frederic-mahe Ah! Thanks for the testing (I don't have an ED myself!) :)
I'll try again on the MegaSG if a new version is made.