SmokeMonsterPacks/Mega-Sg-Jailbreak #11: Game Gear 4:3 width presets incorrect

kingpost, 2019-04-06:
In jailbreak 7.3 using the Game Gear core, the width presets for "4:3 for 16:9" are not correct.

Here are the values they should be for each height preset:

```
Height | 4:3 Width
5.0x      960
5.5x     1056
6.0x     1152
6.5x     1248
7.0x     1344
7.5x     1440
```

frederic-mahe, 2019-04-06:
The Game Gear resolution is 160 ×144. I think you are applying the following formula:

height = 144 * multiplier * 4/3

It seems to me that formula is valid only for the width, not the height, but I might be wrong. Could someone please confirm this?

kingpost, 2019-04-06:
I'm specifically discussing width values, not height.

kingpost, 2019-04-10:
Mr. Mahe, is there any other confirmation you need?  Please note I have reaffirmed the issue is with the "4:3" horizontal/width presets, not the height. Thanks.

frederic-mahe, 2019-04-10:
@kingpost I need another Mega Sg user to confirm your observations so I can tag this issue as a bug. But don't worry, tagged or not, all open issues are considered when preparing a new firmware release.

frederic-mahe, 2019-04-15:
Somewhat related to your issue @kingpost ,FirebrandX has posted a video on that topic. He says that the correct values for Game Gear at 7.0x are  1310 by 1008 px (see https://youtu.be/dzVu05ACVhc?t=2890), corresponding to "square pixels", and the closest you can get to the actual ratio of a Game Gear screen.

Brianvgplayer, 2019-06-16:
Firebrand X later suggested 1344x1008 for GG instead of 1310x1008.

Firebrandx, 2019-11-20:
My original preset was based on someone else's crude measurements of the Game Gear screen. However, I later found the service manual PDF, and the actual dimensions almost perfectly match 960x720 at 5x and 1344x1008 at 7x. So each integer scale should follow that graph. The OP is correct so long as the vertical scale is also honored.

zxtheproto, 2020-09-26:
1428 by 1071, in my experience, fills the screen in the 1.33:1 aspect ratio without cutting off detail.

closed 2023-06-26 by tjanas