SmokeMonsterPacks/Mega-Sg-Jailbreak #12: Alternative Sega CD Bios is not used after reboot

oneoogeoo, 2019-04-07:
After powering off and on the selected alternative Sega CD Bios is still checked in the options, But the Sega CD 2 (european PAL version) starts with its own Bios version.

The only way to use an alternative bios (after powering off and on) again ist to switch to another alternative bios.

TheRealFlamepanther, 2019-04-10:
Same for me on US SCD 2, version 2.00

SmokeMonsterPacks, 2019-04-17:
This should be fixed in v7.4

closed 2019-04-17 by SmokeMonsterPacks

frederic-mahe, 2019-05-09:
same issue with JB 7.4 (see issue #34).

TheRealFlamepanther, 2019-05-09:
The behavior for me in 7.4 is different. If I have the system set to load carts on start up, I still get the stock (hardware) BIOS when I power on. But if I then go to "Run Cartridge" from the menu, I get the custom BIOS.