SmokeMonsterPacks/Mega-Sg-Jailbreak #13: Phantasy Star IV "Rune" theme music plays a sustained monotone sound coming from the Square 3 channel

senseico, 2019-04-08:
About 15 minutes into the game, you meet a character named Rune. Upon first meeting Rune, a specific song plays. During this song, a monotone sound coming from the Square 3 channel is heard. This problem does not occur if the Mega Sg is cold booted just before loading the ROM. It seems to only happen if this is not the first ROM loaded.

Phantasy Star IV (USA).md
MD5: 84cbd0ff47f3c8e9d21d2c2fc39185fa

Firmware JB7.3 2019-04-01

Issue is heard when using default settings. Adjusting the volume of the Square 3 channel to 0 seems to correct this.

I have attached a zip file containing two audio samples: the issue being heard and the issue being corrected by muting the Square 3 channel. The zip file also contains a copy of my save file.

The issue can be reproduced by following these steps with the save file:

1. Load any ROM.
2. Return to the menu and load the Phantasy Star IV ROM.
3. Load save slot 3.
4. Enter the town in front of you and walk north. Speak to Rune by pressing the C button.

[PSIV Samples and Save.zip](https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/files/3055974/PSIV.Samples.and.Save.zip)

Rageypoo, 2019-04-08:
I can confirm this issue. I have also tested with a bugfix patch and it does not correct the issue and is easy to replicate.

senseico, 2019-04-08:
This issue could not be reproduced on a Genesis model 1, even after pressing the reset button to try and simulate how the issue is reproduced on a Mega Sg.

Rageypoo, 2019-04-08:
> This issue could not be reproduced on a Genesis model 1, even after pressing the reset button to try and simulate how the issue is reproduced on a Mega Sg.

were you using real cart or everdrive? the bug is believed to be from rom switching (and possibly loading incorrect values) on the Mega SG only. Reset wouldn't reproduce the issue if it's the same rom.

frederic-mahe, 2019-04-09:
Great issue report @senseico 
Thanks for confirming the issue @Rageypoo

lukeshumard, 2019-04-13:
I can confirm this happens on a real cart with official firmware 4.2. I'll confirm with 4.3 tomorrow.

lukeshumard, 2019-04-13:
Confirmed. This happens on 4.3 as well. Using the real cart reproduces the exact same issue as loading off the SD card.

Rageypoo, 2019-04-13:
Ty for confirming!

senseico, 2019-04-16:
I have confirmed that the issue persists with v7.4.

SmokeMonsterPacks, 2019-04-17:
Is this still present in v7.4?

Rageypoo, 2019-04-17:
> Is this still present in v7.4?

Yes, confirmed in 7.4 just tested it.

Rageypoo, 2019-04-17:
> Thanks!

no problem, if it doesn't happen the first time you play, just reload the rom till it happens, or load another rom then go back, it will for sure happen.

Rageypoo, 2019-04-27:
Any update on this particular bug? Has it been figured out what is causing the audio issues? SoR2 also experiences audio issues and a handful of other games.

Rageypoo, 2019-06-02:
Wow almost 2 months and these bugs are still here?

frederic-mahe, 2019-06-12:
@senseico @Rageypoo @lukeshumard that bug should be fixed in JBv7.5. Could you please confirm?

GreatHierophant, 2019-06-12:
Yes, it is fixed.

closed 2019-06-12 by frederic-mahe

Rageypoo, 2019-06-12:
I was unable to reproduce the bug with JBv7.5