SmokeMonsterPacks/Mega-Sg-Jailbreak #16: Shining Force CD Unable to Save Directly to RAM Cart

FrJosh, 2019-04-11:
This could possibly be an issue with any Sega CD game that saves directly to the RAM Cartridge.  I'm using the db Electronics RAM cartridge and not the official Sega RAM cartridge. It works just fine with games that save only to the internal memory followed up by resetting the Mega SG, going into the Sega CD BIOS, and transferring the save from internal memory to the cartridge.

When starting a new game in Shining Force CD it detects that you have a RAM cartridge inserted, which is then chosen to start a new game, you input the main characters name, then it says "File Access Failed.  Code 1" and just sits there until you reset the console.

Another Mega SG/Sega CD user from Reddit also experienced the same issue when attempting to save to the same RAM cartridge even after starting the original save on an original Genesis.

Firmware version: Jailbreak 7.3
Sega CD Model 2 BIOS 2.00

SmokeMonsterPacks, 2019-04-17:
Is this still present in v7.4?

FrJosh, 2019-04-17:
Just tested, issue still present in 7.4, yes.

GreatHierophant, 2019-04-17:
I had no difficulty saving and loading a game from my Mega EverDrive X7 and its CD RAM capability with Shining Force CD.

FrJosh, 2019-04-17:
How many blocks of memory does the Everdrive give you?  8189?

GreatHierophant, 2019-04-17:
My Shining Force CD save takes 99 blocks.  The Mega EverDrive gies you 2045 free memory blocks after format, compared to 125 memory blocks the base Sega CD gives you.

FrJosh, 2019-04-17:
Do you know of other games that let you save directly to the RAM cart like Shining Force CD does?  I was playing Lunar 1 but that only saves to internal RAM, it can't interface with the RAM cartridge at all.

GreatHierophant, 2019-04-17:
Try Dark Wizard.  Also, what is the revision of your CD RAM cart?

FrJosh, 2019-04-17:
I'll try that out tomorrow.  How do you tell what revision you have?  Open the cart?  Nothing on the StoneAgeGamer label or case says anything about revision.

GreatHierophant, 2019-04-17:
I assume the revision number is on the PCB.  db-electronics mentioned a compatibility issue with Dark Wizard which was fixed in PCB revision v2.0.

Brianvgplayer, 2019-11-25:
Something seems wonky with the RAM cart saving on JB 7.7. After formatting the RAM on the Mega Everdrive X7. I got a message in Sonic CD that the RAM cart was full. After getting the cart to work, I found I had to reformat again. When I saved on the HDG model 1 Genesis, I didn't have this issue.

stardragon2020x, 2020-07-09:
I can verify this is still an issue and I can also verify I am getting the same error with an official SEGA CD RAM Cartridge of which I am the original owner and it still has save files stored on it from my play time several years ago. I thought there was something wrong with my Sega CD console initially until I found this post. Has Analogue addressed this at all? I haven't been able to find much information anywhere else.

tjanas, 2023-05-05:
@stardragon2020x @DeerSteak @Brianvgplayer @FrJosh please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

closed 2023-06-26 by tjanas