SmokeMonsterPacks/Mega-Sg-Jailbreak #18: Audio presets - predefined and custom

Shiftone, 2019-04-12:
Quick feature request.  With everyone's idea of what a "real" Mega Drive should sound like being very subjective, it seems unlikely that we'll arrive at a default that makes most people happy.

How about a set of presets (compiled from the research Firebrand and Ace are doing) that can be selected. If we could also save current audio settings to a custom preset - even better.

For example we could have presets for;

Mk1 Sega Genesis
Mk2 Sega Genesis
Mega Sg Hi Fidelity
Custom preset 1
Custom preset 2

etc etc

TheRealFlamepanther, 2019-04-12:
I would like this. But of course there are differences in how a model 1 or model 2 sounds, depending on the board revision.

SmokeMonsterPacks, 2019-04-13:
We could revisit this in the future once the audio is hashed out. This is an official firmware feature anyways.

closed 2019-04-13 by SmokeMonsterPacks