SmokeMonsterPacks/Mega-Sg-Jailbreak #19: Corrupted in game sprite graphics Mortal Kombat Mega CD

MrFilz, 2019-04-13:
When playing my original pal Mortal Kombat CD on my Mega SG with my pal Sega Mega CD Model 2, the sprites of the characters are glitching out during fights! Tried 1080p60 and 1080p50, pal and ntsc modes! I use the msg_firmware_verJB7.3.bin 
All my other games are working fine, even my pal version of Final Fight CD... But Mortal Kombat is a sprite-mess...  And before you ask, it is working fine with my original Sega Mega Drive!

frederic-mahe, 2019-04-13:
If someone could please confirm that bug, that would be great, thanks.

MrFilz, 2019-04-13:
the issue does only appear with the sega cd (mega cd) version. I have the pal game!

[image]

Maybe it is a pal issue...

SmokeMonsterPacks, 2019-04-13:
Are you using a PAL BIOS?

MrFilz, 2019-04-13:
Yes, the original system. No region free bios. It is the 2.00 System Bios
from my Mega CD 2! Tested the cd in combination with my Mega Drive Model 1
and 2. Without issues!

SmokeMonsterPacks, 2019-04-13:
Thanks. Could you please try forcing that same v2.0 PAL Mega CD2 BIOS through the Jailbreak and see if it's the same?

MrFilz, 2019-04-13:
If i force the bios through the Jailbreak, the game starts with a black screen! Sound is playing, you can listen to the music from the fmv at the beginning, but no picture. Button press is also not registered until the music from the video stops and the game begins. From there everything is unchanged, including the corrupted sprites... :( I tested also the mega cd 1 bios, and a region free modification of the mega cd 2 bios with the same results! I found another mega cd 2 bios file (eu_mcd2_930601.bin). With that it restarts the Mega CD Bios Screen in a loop! 
Tested other Mega CD Games with different bios without issues.

mrploppers, 2019-04-15:
I'm having the same issue with an original NTSC CD on a US Model 2 CD unit. I've tried it with the original bios and using a region free bios loaded on a Mega Everdrive X7

SmokeMonsterPacks, 2019-04-17:
Is this still present in v7.4?

mrploppers, 2019-04-17:
I just tried it again and got the same result.

GreatHierophant, 2019-04-17:
I have also tried Mortal Kombat with the latest firmware and have no issues with character sprites.  I used the PAL CD version, a US CD Model 2 and the Mega-CD 2 (Europe) (v2.00) BIOS.

SmokeMonsterPacks, 2019-04-17:
Thanks, I'm going to close this then.

mrploppers, 2019-04-17:
> 
> 
> I have also tried Mortal Kombat with the latest firmware and have no issues with character sprites. I used the PAL CD version, a US CD Model 2 and the Mega-CD 2 (Europe) (v2.00) BIOS.

How long did you play? It takes a couple matches for me before the corruption shows up. I'm still having the issue.

GreatHierophant, 2019-04-17:
I played at least three matches and ran the game for about 20 minutes.

MrFilz, 2019-04-17:
I do not understand, why this was closed... Nothing changes with the new firmware, the issue is still there with no change. Happens in the original firmware and in the jailbreak! The only thing that changed is, i can now use different bios for the mega cd in combination with the CD version of Mortal Kombat without of any problems like black screen or resetting. But the corrupted sprites are still there!

SmokeMonsterPacks, 2019-04-17:
This was closed because the problem couldn't be replicated. I'll reopen it if it happens to another user.

MrFilz, 2019-04-19:
Sad, but anyway, i will try a us copy of the game on my Mega CD 2 to test if the ntsc version has the same problem with my setup. It is very unlikely, that mrploppers and myself are the only persons out there with that problem. Maybe i should report this to Analogue, cause it also happens  in the original firmware...

alexcom84, 2019-06-10:
I could not replicate this issue using my jpn Mega-CD (fw 4.5) and the NTSC version of the game.

MrFilz, 2019-06-11:
Played thru the game with Sub Zero on official fw 4.5, and the problem did not appear!

closed 2019-06-12 by frederic-mahe