SmokeMonsterPacks/Mega-Sg-Jailbreak #21: NHL '94 Sega CD freezing

mrploppers, 2019-04-17:
When playing NHL '94 CD shortly after the puck drops and the actual game starts the game will freeze and no longer be playable. Sometimes it happens immediately after the puck drops, some times it takes up to a minute to freeze. 

Firmware version: Jailbreak 7.4
Sega CD Model 2 BIOS 2.00

GreatHierophant, 2019-04-17:
This happens to me as well, same specs.

frederic-mahe, 2019-04-17:
@mrploppers @GreatHierophant thanks a lot for your feedback. Was the issue present with Jailbreak 7.3 too?

GreatHierophant, 2019-04-17:
> @mrploppers @GreatHierophant thanks a lot for your feedback. Was the issue present with Jailbreak 7.3 too?

Yes, the same issue appears in JB v7.3, so it isn't a regression.

BooBerry, 2019-06-10:
This was confirmed by Kevtris as a known issue, where the CD's 68K is crashing.

This isn't the only game known to have this issue right now. In addition to **NHL '94**, **Chuck Rock** and **The Adventures Of Willy Beamish** are also known to crash currently. So the issue might need to be edited to reflect that it's not just the one game with this issue. There's likely others that haven't been discovered yet.

frederic-mahe, 2019-06-12:
@mrploppers could you please confirm this is fixed in JBv7.5?

mrploppers, 2019-06-12:
I'll test it tonight or tomorrow night and confirm it.

BooBerry, 2019-06-12:
It's not fixed. Like I said, it's one of three games that remain unfixed due to an issue that's very hard for Kevtris to debug.

BooBerry, 2019-11-19:
This should be fixed in the new jailbreak firmware released today for the three games I listed above (which includes the main issue of this report, NHL '94)... however, Sonic CD is now broken if using a real Sega/Mega CD. The MegaSD is not affected.

closed 2019-11-19 by SmokeMonsterPacks

frgagnon, 2019-12-30:
I had the same issue with Dark Wizard from original CD (the game froze after playing it for a few seconds, the intro played fine and the initial menus as well, but on the map it froze very quickly).

With both the v4.7 firmware and v7.7 JailBreak the problem has been solved for that game as well.