SmokeMonsterPacks/Mega-Sg-Jailbreak #22: Sega CD - Vay - Blue Screen During Intro

FrJosh, 2019-04-17:
## Expected behavior and actual behavior
When starting a new game the screen turns blue during the intro and a short section of the audio (the guy talking to you) keeps repeating over and over.

## Steps to reproduce the problem
Just start a new game and watch the intro

Firmware: 7.3

jairulnait, 2019-04-17:
I experienced the same issue. Model 2 SCD and JB7.4.

GreatHierophant, 2019-04-17:
Ditto

BooBerry, 2019-06-10:
This issue has been fixed in the official 4.5 firmware.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

alexcom84, 2019-06-10:
Tested and working as intended on fw 4.5.

frederic-mahe, 2019-06-12:
It should also be fixed in the jailbreak firmware. @alexcom84 it would be great if you could make the test.

alexcom84, 2019-06-12:
Working well on jb 7.5 !

closed 2019-06-12 by frederic-mahe