SmokeMonsterPacks/Mega-Sg-Jailbreak #23: Sega CD - Popful Mail - MAIN-CPU ERROR !!

FrJosh, 2019-04-17:
## Expected behavior and actual behavior
When you get to the "Press Start" to play main screen, after pressing start the screen goes black with a bunch of text on it.  MAIN-CPU ERROR !!  followed by a bunch of random code.

jairulnait, 2019-04-17:
Again, also experiencing the same issue. It was previously working with JB7.3.

Vorenge, 2019-04-27:
Same issue here. This error occurs after the new game into video. I have the original game and the model 2 Sega CD.

bmichelmd, 2019-05-03:
The first time I ran it I made it past the intro but now it stops at or around the intro with the same CPU error.

mattsolomon419, 2019-05-22:
Same CPU error here. 1st time Sega CD owner. Wanted to play this game real bad.

frederic-mahe, 2019-05-22:
@mattsolomon419 I cannot test it myself, but the game seems to run fine with [jailbreak firmware7.3](https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/blob/master/firmware/msg_firmware_verJB7.3.bin).

BooBerry, 2019-06-10:
This issue has been fixed in the official 4.5 firmware.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

alexcom84, 2019-06-10:
Tested and working great on fw 4.5.

closed 2019-06-12 by frederic-mahe