SmokeMonsterPacks/Mega-Sg-Jailbreak #24: Master System - Alex Kidd in Miracle World, audio bug

vanfanel, 2019-04-17:
In Alex Kidd in MW, when you get to speak with the first henchman before the R/P/S battle, when text is displayed a high-pitch tone is left active after the text is printed.

This does not happen on real hw (I know the game very well as this was included in every Master System II here in europe).

GreatHierophant, 2019-04-18:
Yes, the tone is not present on my Model 1 Genesis in SMS mode when you face the guy who makes you play the rock, paper, scissors battle.

frederic-mahe, 2019-04-18:
is the audio bug present when using an original cartridge with the Mega Sg? (assuming you've tested with a ROM and an Everdrive so far)

vanfanel, 2019-04-18:
@frederic-mahe : I dont have a phisical cartidge for it. It is included on my Master System ROM memory.
I run it on the MegaSG via JB.

@GreatHierophant : Do you have a phisical cartidge to test?

GreatHierophant, 2019-04-19:
No, but ironically the tone is not present on the earlier incarnation of this core found on the Nt Mini.  It's clearly a bug in the Mega Sg.

skooterblog, 2019-04-27:
Here is the bug on video:
https://youtu.be/_lN6pHk4-hY?t=423

Running the BIOS version through the jailbreak firmware.

If I recall correctly, the does not exist on my real Master System II (Tec Toy), which has this BIOS.
It also does not exist on my real Sega Genesis running the hacked BIOS through Mega Everdrive.

BooBerry, 2019-06-10:
According to the changelog, this should be fixed in the official 4.5 firmware now.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

frederic-mahe, 2019-06-12:
@vanfanel could you please confirm this is fixed in JBv7.5?

GreatHierophant, 2019-06-12:
I can confirm that it is fixed.

closed 2019-06-14 by frederic-mahe