SmokeMonsterPacks/Mega-Sg-Jailbreak #25: Pier Solar red screen with model 1 Sega CD

mario-64, 2019-04-20:
I own a genuine Pier Solar cartridge from the reprint run they did. If I boot with the Mega SG connected to my model 1 Sega CD I can’t get past the Watermelon logo screen. It goes straight to a red screen. Pressing the reset button doesn’t help. It does the same thing still. If I disconnect the Sega CD, the game boots normally. 

I tested using the Rev A, B and C rom files and they will load but the game does not recognize that the Sega CD is connected - therefore the soundtrack disc cannot be used. This is with JB 7.4 firmware.

Expected behavior: Genuine game cartridge should load correctly when Mega SG is connected to model 1 Sega CD. Rom files should recognize when a Sega CD is connected

Actual behavior: Genuine game cartridge hangs at a red screen following the Watermelon splash screen when Mega SG is connected to model 1 Sega CD. Rom files do not recognize a connected Sega CD

Official 4.4 firmware has the same problem. Tested with and without cartridge audio enabled in SG settings

mathewbeall, 2019-04-30:
This happens with the model 2 sega CD as well.

mario-64, 2019-05-05:
Can we remove the Needs Verification label now given mathewbeall’s report?

BooBerry, 2019-06-10:
This has been listed to be working in the official 4.5 firmware, and Kevtris has said he tried this on a model 1 Sega CD, model 1 Japanese Mega CD and model 2 Sega CD and it works fine. I'd give 4.5 a try and see if this is still happening.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

mario-64, 2019-06-10:
Unfortunately 4.5 does NOT fix this issue. Just tested. There are multiple revisions of Pier Solar. I wonder if perhaps Kevtris tested with the original release not the reprint.

frederic-mahe, 2019-06-10:
thanks @mario-64 is there a revision number attached to _Pier Solar_ we could use to sort out that issue?

BooBerry, 2019-06-10:
Pictures of the PCB of the cart probably wouldn't hurt either.

mario-64, 2019-06-10:
[image]
[image]

Attached are the requested pictures

mathewbeall, 2019-06-11:
Confirmed - still not working for me, and I have the exact same Rev

mario-64, 2019-06-11:
I offered to send Kevtris my cart for testing but as yet he hasn't responded :-(

mario-64, 2019-06-12:
Does the new jailbreak firmware allow loaded rom files to recognize that a Sega CD is connected? That would at least allow us to play the game with CD audio. Using the previous firmware, Pier Solar did not detect a connected SCD

Xaranar, 2019-12-24:
Still an issue as of firmware 4.7.  As an aside, I had this issue on real hardware on certain models of Mega Drive.  No idea what caused it.

tjanas, 2023-05-05:
@mario-64 please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

BigPrimeNumbers, 2023-05-22:
> @mario-64 please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

Can confirm still does not work. Goes to red screen on initial boot, and hangs on opening text if reset from menu.

tjanas, 2023-06-25:
@BigPrimeNumbers Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

closed 2023-06-26 by tjanas

BigPrimeNumbers, 2023-06-26:
> @BigPrimeNumbers Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

I'm away right now and can't test until late next week; but I'll try it out when I get back!

BigPrimeNumbers, 2023-07-03:
> @BigPrimeNumbers Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

Just to close this loop, tested, and it is still broken on jbv7.9. Opened an issue over at https://github.com/analoguejb/Analogue-Mega-Sg-JB.