SmokeMonsterPacks/Mega-Sg-Jailbreak #26: Video corruption on JapaneseMega-CD BIOS screen and some games

ChronoStriker1, 2019-04-22:
## Expected behavior and actual behavior
Not have graphical corruption

## Steps to reproduce the problem
Starting Mega-CD while attached to the Mega-SG - tested with all Bios's named in smokemonster Mega Everdrive pack, region free and original.  All have issues especially in the memory management screens. Also tested loading via the Mega-SG and via Mega Everdrive X5.  The Mega-CD acts normal with no video corruption in BIOS or games when connected to a US Model 2 Genesis loading the BIOS via the Everdrive.

## ROM name and hash value
BIOS, Micky Mania, Earthworm Jim Special Edition

If you can, please indicate if the bug occurs when using an original
cartridge as well.

## firmware version
Jailbreak 7.4 - Only tested with this version as I just received the Mega-CD

Examples of the issue
https://imgur.com/a/VgNTQFa

Nigoli, 2019-04-23:
My Mega CD M1 was like this on the BIOS screen. Had to send it in for repair because it was a capacitor problem.

I would play on the Genesis M2 with it for a couple of hours and see if the problem happens on it like it was on the Mega Sg.

ChronoStriker1, 2019-04-23:
When checking the capacitors appear fine.  While I'm not against replacing them the fact it works fine with the normal Genesis implys that the mega-sg is doing something different which I would assume it shouldn't.

alexcom84, 2019-04-24:
Well I have the Mega CD JPN launch model and only got the BIOS background with some white noise. Thought it was a cap issue so I grabbed a kit from console5.com and recapped all 3 PCBs (EXT board pcb on this model has 3 caps and a connector that goes on the oscillator daughterboard) and got the same issue. Checked all traces from 68k to RAM and ASIC w/o problems.
 Gave up and grabbed another Mega-CD JPN launch model that the vendor send me video with it working.... and hey surprise it has the same issue.
 Memory management screen is all black with some garbled graphics. Lords of thunder, Popful Mail, Lunar E.B., Snatcher and Sonic CD RAM management screen have garbled graphics.
 On JB 7.4 fw and tried all bios roms for model 1.

alexcom84, 2019-04-30:
I think the problem with the Mega-CD 1 might be the fact that the jpn launch model uses 4x 4bit DRAMs ( https://images.app.goo.gl/fNQJWb38y6HAFeXr9 )
and the export revisions had a single 16bit DRAM 
https://images.app.goo.gl/mGNrWjFw9xE64uDc8

The EXT port pcb is different as well for this model:
http://wellby.dyndns.org/dnl/img/con-bd5.jpg?
http://wellby.dyndns.org/dnl/img/con-bd4.jpg?

alexcom84, 2019-05-01:
https://twitter.com/maxx_2nd/status/1117106346384433152?s=21

Japanese user experiencing the same issues with Mega-CD 1.

BooBerry, 2019-06-10:
This should be fixed in the official 4.5 firmware now.

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

alexcom84, 2019-06-10:
Yup! Everything is working great now. 😃

ChronoStriker1, 2019-06-10:
Confirmed that it fixed BIOS vid, Micky, and Earthworm Jim for me but it broke Lunar.  Tried multiple bios and all have the same corruption at the start of the intro video, sound is fine though.  Video attached.

https://drive.google.com/open?id=1QuSs3f_mvPwppTqdcAgqfia7vEf2rYXf

alexcom84, 2019-06-10:
Tested both Lunars on my model 1 without any problem. Even the uncensored ones with mixed case font.

ChronoStriker1, 2019-06-12:
I also tested with a Japanese copy of Lunar and it is showing the same symptoms.

ChronoStriker1, 2019-06-12:
And just to be through re-flashing 7.4 allows the game to start normally.

alexcom84, 2019-06-12:
At this point since this issue is no longer regards the JPN Mega CD bios but only certain games on the model 2 Mega-CD, you should probably either modify the title or close the issue and have @ChronoStriker1 open a new one with the list of games that are not working on his model 2 unit.

ChronoStriker1, 2019-06-13:
As BIOS issue appears to be resolved I will open a separate ticket in regards to the game corruption issue.

closed 2019-06-13 by ChronoStriker1