SmokeMonsterPacks/Mega-Sg-Jailbreak #27: Rockman Mega World & Mega Man The Wily Wars EEPROM Rom save problem

Nigoli, 2019-04-23:
## Expected behavior and actual behavior
EEPROM type on the Rom of Rockman Mega World & Mega Man The Wily Wars fails to save. I'm able to test this with the Alt version of Rockman Mega World (normal one is already SRAM Patched according to the Mega EverDrive X7).

Mega Man The Wily Wars has the same issue.

## Steps to reproduce the problem
Defeat Gutsman in Rockman 1, and try to save, go back into the Mega Sg file browser, then reload the game.

## ROM name
Rockman - Mega World (Japan) (Alt 1).md
Mega Man The Wily Wars (E).md

## firmware version
JB v7.4

Rageypoo, 2019-04-26:
I am unable to reproduce this bug

Nigoli, 2019-04-27:
Even with those ones listed above?

Why won’t it save for me? Greatest Heavyweights (U).md was like this and that saved fine too. It’s also the same serial EEPROM too.

SmokeMonsterPacks, 2019-04-27:
My guess is that your ROM is pre-patched @Nigoli

Rageypoo, 2019-04-27:
Are you a part of the Classic Gaming discord? If so you can message me on there, same name as here, perhaps I can work through it with you and we can figure it out.

Nigoli, 2019-04-27:
I’ll have to message you. The Japanese version that I wanted to use definitely wasn’t working with the Alt 1 version. The Mega EverDrive X7 detected it was an EEPROM type.

Rageypoo, 2019-04-27:
> I’ll have to message you. The Japanese version that I wanted to use definitely wasn’t working with the Alt 1 version. The Mega EverDrive X7 detected it was an EEPROM type.

ok sounds good, I wont be able to test with an everdrive as I do not own one, but perhaps we can see what's going on with the jailbreak itself.

Rageypoo, 2019-04-27:
@SmokeMonsterPacks I was able to reproduce the bug.

Rageypoo, 2019-04-27:
Looking within the save folder, it seems the game does make a save file but it's okly 1kb while a patched rom (or any other roms really) seem to be 128kb.

Nigoli, 2019-04-27:
[save.zip](https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/files/3123523/save.zip)

This is the file that it produces, and the EEP is what the Mega EverDrive X7 produces with it's EEPROM support if needed for comparison.

Nigoli, 2019-04-27:
I restested with different dumps. I can definitely say that my theory is right. The one from "Unpatched EEPROM-Save Games for X7" was most definitely SRAM patched with a SHA-1 & MD-5 _(7a25e5249c0a406f20f76bef45dc3202 4dfcf5c07106f6db4d8b7e73a053bb46a21686c3)_.

Ones from normal sets, "Rockman - Mega World (Japan).md") are unpatched without SRAM modifications and it still won't let me save correctly. I also did a hex match with the Alt 1 version and the normal one and they are the same in both SHA-1 & MD-5.

@SmokeMonsterPacks Can this please be reopened? I won't be able to do anymore testing until a CopyGenesis is implemented or if I ever buy a Retrode 2 for dumping my original cartridge for the Japanese version.

Nigoli, 2019-04-28:
Thank you @SmokeMonsterPacks 

Unrelated to the issue. I did some more research on this game. Apparently there are two types of cartridges manufactured for the Japanese release of the game. I'm not sure how true this is. But it could clear the confusion though and make more sense why there's two types of these dumps labeled the way they are. Apparently, earlier/earliest prints of the game seem to use SRAM type saving, while all later editions seem to be EEPROM type. I'm pretty certain that my copy of the game was EEPROM type because it seemed to have save data retained when I booted it up.

Maybe this is why the "Alt 1" is labeled as such because that had EEPROM type. Some of them are mislabeled as just being Rockman - Mega World from the start either being one or the other, SRAM and EEPROM. I think it remains to be seen or actual confirmation how true the SRAM type cart existence is to begin with. It's hard to know because the game shares the same game serial code. The international release has and always has been EEPROM type.

Rageypoo, 2019-04-28:
> **Update:** http://gendev.spritesmind.net/forum/viewtopic.php?p=2485#p2485 <- This is where I got the information from. I wish I could find more info on this.

Thank you for that link, I had no idea how many games used this, NBA Jam was interesting to see.

Nigoli, 2019-06-12:
After testing with Firmware JB7.5, both regions of the game save perfectly!

Thank you Kevtris for fixing this.
Thank you SmokeMonster for reopening it for me.

closed 2019-06-12 by Nigoli

spotanjo3, 2021-02-01:
I would like to know what is checksum for those two games that solved save problem, please ?