SmokeMonsterPacks/Mega-Sg-Jailbreak #29: Video corruption in Wings of Wor (and Gynoug)

Rageypoo, 2019-04-25:
At the start of the game, within a minute and a half, the screen shakes similar to mode 7 on snes. On a real genesis, the full screen shakes without issue, on the mega SG the left side of the screen experiences video corruption. 

Here is a link to a playthrough where there is no video corruption (Expected behavior) https://youtu.be/sjiGztXBOYE?t=71

You can reproduce the problem simply by playing up to this point on the first stage. It does occur again far later in the stage and happens again there. 

Wings of Wor (U) [!].gen
Gynoug (E) [!].gen
Gynoug (J) [!].gen

On JB 7.4 with 1080p fill settings, tested on x4 and x5 as well, with interpolation both on and off.

frederic-mahe, 2019-04-26:
I was able to reproduce this bug with the stock firmware, and using a [valid ROM](https://datomatic.no-intro.org/?page=show_record&s=32&n=1537):
```
Wings of Wor (USA).md 12962a34ddb87886f715903df711c5838984f6a109c2d54ec1d63a92c6df8c2a 91caad60355f6bd71949118bd30ea16d0f4c066e	911ef3d3909b19609f76f0d70a017d85	210a2fcd
```

The-Iron-Goat, 2019-04-26:
My SG does the same thing with firmware 4.4, a MED v1, and the above ROM. This same glitch occurs on emulators as well (tested on MAME and BlastEm). Glitch does not occur on original console (tested with a Model 3 Genesis and the same MED and ROM).

Rageypoo, 2019-04-26:
> My SG does the same thing with firmware 4.4, a MED v1, and the above ROM. This same glitch occurs on emulators as well (tested on MAME and BlastEm). Glitch does not occur on original console (tested with a Model 3 Genesis and the same MED and ROM).

thank you for testing!

jonasongithub, 2019-04-26:
iirc when this was reported on the classic gaming discord, people reported, that the bug was also present on original model 1 and 2 genesis. So maybe this game only runs properly on the model 3..

The-Iron-Goat, 2019-04-27:
Update: the Higan and Genecyst emulators do this effect correctly. Yes, Genecyst, the DOS-based emulator from 1998.

Rageypoo, 2019-04-27:
> Update: the Higan and Genecyst emulators do this effect correctly. Yes, Genecyst, the DOS-based emulator from 1998.

Higan does what analoguedont

SmokeMonsterPacks, 2019-04-27:
If the game glitches on real Model 1 & Model 2s, and only runs correctly on the horribly crappy Model 3's, then the Mega Sg is reproducing a 1/2 exactly.

The-Iron-Goat, 2019-04-27:
Update: the Higan and Genecyst emulators do this effect correctly. Yes, Genecyst, the DOS-based emulator from 1998. 

Edit: Had a Model 2 Genesis handy and tried it, and the glitch is there. Went back to Model 3 and didn't see it. The "glitch" might be an accurate representation of original hardware, at least with respect to the model 1s and 2s. Wonder if this had to do with bugfixes when Sega switched to the ASIC "Genesis on a Chip" that integrated both processors, both sound chips, and the VDP into a single package. The GOAC first appeared on Model 2 VA4 boards. Hold on, let me see if I have one around here...

The-Iron-Goat, 2019-04-27:
Yup, Model 2 VA4 is good, so I'm guessing the GOAC fixes whatever was going on there.

Rageypoo, 2019-04-27:
> Yup, Model 2 VA4 is good, so I'm guessing the GOAC fixes whatever was going on there.

So then it is a problem with the Mega SG, thank you for confirming.

Rageypoo, 2019-04-27:
Please re-open this issue please.

Rageypoo, 2019-04-27:
@SmokeMonsterPacks please re-open issue

frederic-mahe, 2019-04-27:
The glitch is present when playing on a Model 1 Genesis VA6 (US), using a MegaEverdrive X7.

Like @The-Iron-Goat said the Mega Sg is an accurate representation of the original hardware (with discrete sound chips, and VDP). This is similar to the Super Nt situation: kevtris chose to replicate the behavior of the most common Super Nintendo models, and not the behavior of later less expensive redesigns such as 1-CHIPs and Junior.

I vote to close that issue.

jonasongithub, 2019-04-27:
It definitely doesn't seem to be a Mega SG bug, but rather a bug of the game itself, not being perfectly compatible with the earlier revisions of the console. Maybe this could be changed into a feature request instead of a bug report.

frederic-mahe, 2019-04-27:
This is an interesting finding. Either someone fixes the game to work on Models 1 and 2, or writes a GOAC Model 3 FPGA core. Both these options are outside the scope of the jailbreak firmware, so I am going to close that issue.

closed 2019-04-27 by frederic-mahe

Rageypoo, 2019-04-27:
> This is an interesting finding. Either someone fixes the game to work on Models 1 and 2, or writes a GOAC Model 3 FPGA core. Both these options are outside the scope of the jailbreak firmware, so I am going to close that issue.

If the model 2 has a fix for these issues it seems intended and should be investigated further, this is incredibly dishonest to dismiss it without actually looking into it.

frederic-mahe, 2019-04-27:
@Rageypoo this issue tracker is for jailbreak-specific bugs, and I am merely trying to maintain that focus.

Closing an issue does not mean it has been forgotten: it is still visible, you can add comments, you can link to that page, and it can be re-opened anytime, if need be. What you've reported is interesting and worthy of further investigation. I invite you to discuss it with the people at Analogue (we are not Analogue here), and with the retro-gaming community in general.

Rageypoo, 2019-04-27:
> @Rageypoo this issue tracker is for jailbreak-specific bugs, and I am merely trying to maintain that focus.
>
> Closing an issue does not mean it has been forgotten: it is still visible, you can add [...]

Fair enough, thank you for the reply.