SmokeMonsterPacks/Mega-Sg-Jailbreak #31: Sound problem in Aladdin

irianordaz, 2019-04-28:
Sound problem in the bonus roulette game at the end of each stage in Aladdin. The sound is not only incorrect but it seems to cut in and out randomly.

I am using v7.4 jailbreak with FirebrandX cleanest sound settings below. The ROM is from SmokeMonster Mega Everdrive pack.

Cleanest Possible Sound:
Ladder Effect Depth = 0
YM2612 High Quality Audio Mode = checked
No LPF
YML = 100
YMR = 100
Square 1 = 32
Square 2 = 32
Square 3 = 32
Square 4 = 29

Rageypoo, 2019-04-29:
Can you provide an example of what the expected sound is and what the actual sound is? Also provide the ROM version name (for example aladdin(b1).gen or something to that effect) and are you using an everdrive to run it or directly from the jailbreak?

irianordaz, 2019-04-29:
The ROM is from Everdrive Pack v4.4 and it is simply called "Aladdin (USA).md". I was initially experiencing the problem when loading the game from the SD card. That said, it does not look like it is a ROM problem. 

I ran my original (official) Aladdin cartridge on the Mega Sg and experienced the same missing sound of the ticking roulette, although the sound does pop in momentarily when you press a button and then fades away again. 

I also tried the same ROM in question through the Mega Everdrive on an original model 1 Sega Genesis and that exhibited no sound issue so the problem is specific to the Mega Sg.

I recorded two videos showing the correct sound and the missing sound but I am unable to attach anything larger than 10M.

Rageypoo, 2019-04-29:
If you happen to have a youtube u can link it, but I'll try to recreate it with the info u gave me. I also sent you a direct message on discord to test your rom.

frederic-mahe, 2019-05-03:
Thanks @Rageypoo for helping @irianordaz to isolate and reproduce the bug. It would be ideal if you could compare the checksum of your ROM with the No-Intro database (see https://datomatic.no-intro.org/?page=show_record&s=32&n=0088).

frederic-mahe, 2019-05-03:
For future reference, the _Aladdin_ ROM listed in SmokeMonster's pack matches [No-Intro's checksums](https://datomatic.no-intro.org/?page=show_record&s=32&n=0088):
```
Aladdin (USA).md
a3779fc77994780e80d05bb557f800110d0398d34b951baa8c0a14910014ded3
d21c085b8429edc2c5092cd74ef3c36d01bf987f
a6fc42bd5d19e36e5d8a2d8c4f4e9d23
ed427ea9
```

frederic-mahe, 2019-05-03:
@irianordaz if you could confirm this is the ROM you've been using, that would be great.

On Mac and Linux, use command line tools to get the hash values of your ROM (for example: `sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility.

Rageypoo, 2019-05-03:
> Thanks @Rageypoo for helping @irianordaz to isolate and reproduce the bug. It would be ideal if you could compare the checksum of your ROM with the No-Intro database (see https://datomatic.no-intro.org/?page=show_record&s=32&n=0088).

no problem, I have tried reaching out to him through discord but I have not received a response yet, when I do hopefully I can figure out which ROM he is using and update here for you, i'll also see if I can reproduce the bug with other versions.

Rageypoo, 2019-05-06:
@frederic-mahe I was unable to reach him and I have left the Classic Gaming Discord. Hopefully someone else can verify it for you.

irianordaz, 2019-05-07:
Hi. Sorry I've been swamped here lately and I've had no time to follow this. I will look into uploading the sample videos next weekend. That said, I posted a comment earlier stating that I ended up testing my original Aladdin cartridge and that exhibits the same problem so it is not a ROM problem.

BooBerry, 2019-06-10:
This has been fixed in the official 4.5 firmware.

frederic-mahe, 2019-06-12:
@irianordaz could you please confirm this is fixed in JBv7.5?

Rageypoo, 2019-06-12:
I can confirm it is fixed in JBv7.5?

BooBerry, 2019-06-12:
I can confirm it's fixed too.

irianordaz, 2019-06-12:
Not sure why it is being confirmed by others by I am still getting the exact same problem as before with JBv7.5 using the same sound settings as I stated above. Those confirming that this is fixed, can you post your settings so that I can test them?

Here are some videos of the sound effect issue. I am very closely familiar with this game, and although it is a small sound effect but may be a problem in other games.

Below is what the roulette should sound like, this is recorded from an original model 1 Sega Genesis:
[](https://www.dropbox.com/s/7l8s6msulit0rnq/Correct.MOV?dl=0)

Below is the missing/broken roulette sound using the MegaSG
[](https://www.dropbox.com/s/efo0kwtxhei148i/Problem.MOV?dl=0)

Hopefully the links work.

BooBerry, 2019-06-12:
I'm using the Pre VA7 Genesis Model 1 settings found here.

http://www.firebrandx.com/downloads/Mega%20Sg%20Audio%20Settings%20for%20the%20Obsessive%20Compulsive%20(Jailbreak%20Edition).txt

And it sounds fine to me now after Kevtris fixed the PSG issue.

Rageypoo, 2019-06-12:
As we stated before, we have no idea which rom you are using, and so we are using any rom with JB7.5 and I am not experiencing any sound problems. I do not own real cart to test.

BooBerry, 2019-06-12:
> As we stated before, we have no idea which rom you are using, and so we are using any rom with JB7.5 and I am not experiencing any sound problems. I do not own real cart to test.

@irianordaz said the Smokemonster Mega Everdrive pack is being used. So it's either **Aladdin (USA).md** or **Aladdin (Europe).md** or **Aladdin (Japan).md** likely being used here. I tested **Aladdin (USA)** using a Mega Everdrive X7.

irianordaz, 2019-06-12:
I have stated before in my second post, this was tested with a real cartridge, it has absolutely nothing to do with the ROM but no one seems to be listening. Quite frankly, I don't care anymore, if everyone is satisfied that it is fixed then close away. I will not be responding to this thread any longer.

BooBerry, 2019-06-12:
@irianordaz in your videos, are you talking about the ticking sound (which indicates movement for the items on the roulette game) and not the sound if you correctly choose an apple/extra life?

Rageypoo, 2019-06-12:
> I have stated before in my **second post**, this was tested with a **real cartridge**, it has absolutely nothing to do with the ROM but no one seems to be listening.

> The ROM is from **Everdrive Pack v4.4** and it is simply called "**Aladdin (USA).md**". I was initially experiencing the problem when loading the game from the SD card. That said, it does not look like it is a ROM problem.
> 
> I ran my original (official) Aladdin cartridge on the Mega Sg and experienced the same missing sound 
> 
> I also tried the same ROM in question through the Mega Everdrive on an original model 1 Sega Genesis 

I mean, your second post said it was a ROM file u tried, and then later tried a real cart. I've tested the ROM from smokemonster pack with JB7.5 and experienced no sound issues, using clean sound settings from FirebrandX. 

I do not own a real cart so that's as far as I can go. I did reach out to you through discord and asked if you could send me the copy of your ROM to make absolutely sure it was reproducable but you never replied. 

¯\_(ツ)_/¯

frederic-mahe, 2019-06-13:
@irianordaz This sound difference could be limited to original cartridges. I am reopening this issue to try to gather more info.

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks