SmokeMonsterPacks/Mega-Sg-Jailbreak #33: Save backup for real catridges

thiagofacanha, 2019-05-07:
(please note that feature requests will be marked as **closed** after a couple days to maintain the focus on unresolved bugs)

**Is your feature request related to a problem? Please describe.**
I have some rpg cartridges and my great fear is the day when Im gonna loose my save because of the death of the Battery. If I change the battery I will loose too the game 

**Describe the solution you'd like**
Would be great if Mega SG was able to create a backup of a save inside a cartridge and if possible being able to "upload" this save to a cartridge again(after replacing a battery for example).

Thxs!

frederic-mahe, 2019-05-15:
Noted

closed 2019-05-15 by frederic-mahe

thiagofacanha, 2019-05-15:
So. Is it possible? I read that there is a cartridge adapter (Retrode)  capable of backup creation but coudn't confirm if it was able to write this backup in a cartridge.
Thxs!

thiagofacanha, 2019-05-15:
Sorry...  found the manual(https://www.retrode.com/uploads/Retrode2-Manual-April2012.pdf) Its possible to write back to cartridge
" These files are by default write-protected. In order to save your progress back to the cartridge, disable the write protection in the configuration file."
Would be possible to do this in the Mega SG?

frederic-mahe, 2019-05-15:
I cannot comment on hypothetical features, sorry.