SmokeMonsterPacks/Mega-Sg-Jailbreak #34: Mega Sg on JB7.4 will not load alternative CD bios file, after saving settings and rebooting the system.

dododub, 2019-05-08:
**Describe the bug**
A selected alternative bios for the Mega CD does not load again after setting one and saving the system settings of the Mega Sg, and then turning the Mega Sg off and on. Even though the "Use replacement CD BIOS" option is checked and the selected BIOS is shown on screen, you have to reselect the bios file you want to load again for it to work.

Auto region detection is turned on in the hardware menu, as required.
Bug shows up on a Mega Sg in combination with a PAL Mega CD 2.

**To Reproduce**
Steps to reproduce the behavior:
0. Make sure the auto region detection is turned on in the hardware menu
1. Check the option to use a replacement CD BIOS 
2. Select a replacement BIOS loaded on your SD card
3. Save your settings.
4. Turn off the Mega Sg
5. Turn the Mega Sg back on
6. See the alternative CD bios settings in the Core Options menu
7. You should see that the box for using an alternative bios is checked, and you should see the path and filename of the alternative bios you chose and saved
8. Go back to the main menu and select "Run Cartridge"
9. The Mega CD will start but will still be running its default system BIOS instead of the alternative.
10. By selecting the alternative BIOS file again from the SD card and then starting the Mega CD again through "Run cartridge" the alternative BIOS will be used.

**Expected behavior**
Expected behavior is that the alternative BIOS should load that was previously selected and saved, instead of the default Mega CD BIOS.

**Firmware version**
Only submit reports for the latest jailbreak or official firmware (please be ready to retest your issue if the firmware is updated).

**Relevant Mega Sg settings**
JB 7.4

**ROM name and hash values**
Different bios files have been tried, but this specific report was made by using bios called "US Sega CD 2 (Region Free) 930601 l_oliveira.bin"

alexcom84, 2019-05-09:
Does the same thing on the JPN model but wanted to wait until this unit was actually supported before reporting other bugs.
  To make it use the replacement BIOS you either select the bios again or run a game from the SD card first, then select “Run cartridge”. 
 Renaming it to cdbios.bin and placing it in /BIOS/ cdbios.bin won’t make any difference.

MP2E, 2019-05-09:
Confirmed here on US Sega CD model 2 as well

alexcom84, 2019-05-09:
At this point maybe it’s better to reopen the old issue from jb 7.3 instead of having a new one.

frederic-mahe, 2019-05-09:
Linking to issue #12 should be enough.

Nicklong79, 2019-05-12:
I have experienced this bug, there is a very simple workaround. When you start up your system, go to cores first, select genesis then back out to the root menu again. Now select run cartridge and your Sega/Mega CD will start up with the bios you have selected. Enjoy.

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks

djspringate, 2023-03-11:
@SmokeMonsterPacks This isn't fixed. I've been having this issue on 7.8 with a PAL Mega CD1, using a Japanese MCD bios. Works but SG loses the settings after reboot. Retains the settings if only pressing Reset but not on cold boot.