SmokeMonsterPacks/Mega-Sg-Jailbreak #35: Audio Glitches and Freezing in Brave Battle Saga (Piko Interactive cartridge)

targrus, 2019-06-04:
Audio distortion in intro sequence and during combat, freezing if left idle in the first forest area.

**To Reproduce**
Boot the game, roll intro. Play until first forest area/dungeon location.

**Expected behavior**
Clear audio, no freezing.

**Firmware version**
7.4

**Additional context**
I'm willing to provide a copy of the cart for debugging.

targrus, 2019-06-04:
Sorry, forgot to mention. This behavior does not happen on a Sega Nomad.

frederic-mahe, 2019-06-05:
Thanks @targrus I assume the game was tested on Model 1 or 2 Megadrive/Genesis too. Let's see if someone else can confirm that glitch, so we can remove the `needs verification` tag.

targrus, 2019-06-05:
Unfortunately I only have access to a Nomad for comparison. Hopefully someone else can verify.

BooBerry, 2019-06-10:
Does this still happen in the official 4.5 firmware that was just released?

https://support.analogue.co/hc/en-us/articles/360024836892-Mega-Sg-Firmware-Update-v4-5

targrus, 2019-06-10:
I just tried this out on the 4.5 official firmware and cannot reproduce the issue. I'm guessing in fixing other sound bugs this fixed this as well. The sound matches the Nomad and no freezing occurs. Awesome!

targrus, 2019-06-12:
I can confirm that it's also fixed in the 4.5 jailbreak firmware. Awesome job!

closed 2019-06-12 by targrus