SmokeMonsterPacks/Mega-Sg-Jailbreak #36: Mega Drive/ Genesis games won't load

neojuk, 2019-06-06:
Has anyone had this issue/got any advice?

I applied the firmware, it worked/rebooted. I tested Master System games, these are all fine. I tested Game Gear games, these are all fine. (I haven't tried any Coleco stuff)

Mega Drive games just straight do not work for some reason though. I try to load any game, I've tried probably 50 different Mega Drive games from Japan, Europe, USA and not one of them will load. What I get on both firmware 7.3 and 7.4 is just a black screen with random white chunky pixels all over the screen.

I've tried 3 different brands of SD cards. SanDisk (4GB), Toshiba (2GB), Kingston (2GB). All standard size SD cards, no Micro SD.

I'm out of SD cards to try now and struggling to see where I'm going wrong. I used the same Kingston (2GB) SD card for the Super Nt firmware and that wasn't an issue. I just tested it again now and all the SNES/SFC ROM files are loading up no problem on the Super Nt with that card.

Any help or advice with this would be much appreciated. Super frustrating this type of thing when it's not working. One last thing the Mega Drive ROM files do load up to 100% before I get the weird screen. All SD cards I've tried are formatted to FAT32 also.

I have attached a picture of this. This is what I get every time:
[image]

neojuk, 2019-06-07:
EDIT: Today I bought 3 more SD cards, one being a fairly expensive one at that, and still this exact same screen every time. Absolute garbage.

neojuk, 2019-06-07:
I am an idiot. I just bought a bunch more SD cards, some fairly expensive too, and I still had the same problem. I had one more look at the files on my computer and noticed that every one of these ROM files has the .smd extension. I changed all the files to .gen files and every one of them works. Man do I feel stupid. Extremely stupid. Go on, point and laugh, it's fine.

closed 2019-06-07 by neojuk

frederic-mahe, 2019-06-07:
@neojuk good you've find what was causing the issue. I was worried you had a hardware issue.