SmokeMonsterPacks/Mega-Sg-Jailbreak #38: Certain audio options have been removed in Coleco core via JB.7.5

SegaSnatcher, 2019-06-11:
This core no longer has the options to apply advanced filter settings, swap left and right audio, or cartridge volume adjust.  There was no mention of this downgrade, so perhaps it was a mistake by the writer of this firmware.

mholzinger, 2019-11-19:
We could close this issue. The Coleco audio options are in the latest 7.7 jailbreak.

Headphone Settings: 
- Volume[0 - 50]
- High Impedence phones [enable/disable]

Channel Levels: [all defaulted to 60]
- Square 1 [0-100]
- Square 2 [0-100]
- Square 3 [0-100]
- Noise [0-100]
- SGM 1 [0-100]
- SGM 2 [0-100]
- SGM 3 [0-100]

Channel panning: [-100 (left) + 100 (right)] (defaults shown below)
- Square 1 [-32]
- Square 2 [+32]
- Square 3 [0]
- Noise [0]
- SGM 1 [-32]
- SGM 2 [+32]
- SGM 3 [0]

-3db output cut [enable/disable]

SmokeMonsterPacks, 2019-11-20:
Excellent, thank you @mholzinger!

SegaSnatcher, 2019-11-20:
Hi,  sorry, must have confused my original post.   I was referring to Low Pass Filter options and left/right swap.  Those use to be in the Coleco core and then were removed and are still gone in curent JB.

tjanas, 2023-06-25:
Per JB7.9 changelog:
Coleco's filter option menu is back.

closed 2023-06-25 by tjanas