SmokeMonsterPacks/Mega-Sg-Jailbreak #39: Not able to load alternate Mega-CD firmware on 7.5

ChronoStriker1, 2019-06-12:
**Describe the bug**
Unable to load alternative Mega-CD bios

**To Reproduce**
Steps to reproduce the behavior:
    Make sure the auto region detection is turned on in the hardware menu
    Check the option to use a replacement CD BIOS
    Select a replacement BIOS loaded on your SD card
    Save your settings.
    Run cartridge

**Expected behavior**
Mega-CD loads with alternative bios

**Firmware version**
JB7.5

frederic-mahe, 2019-06-12:
@ChronoStriker1 do you confirm that you were able to load the same alternative bios with JBv7.4?

ChronoStriker1, 2019-06-12:
Yes, and I can confirm if I downgrade back to JBv7.4 the settings works as expected.

alexcom84, 2019-06-12:
I confirm that loading an alternate bios won’t work on my model 1. If someone could test on their SegaCD to rule out it is not affecting only the JPN units.

frederic-mahe, 2019-06-12:
is this linked to issue #34 ?

alexcom84, 2019-06-12:
Another user confirmed this issue on the atariage thread on a NA model 2, so it’s not related only to the jpn units.
  
It’s kinda different because the alternate bios would not run at all, where in previous issue the problem was only happening on next power on. As someone else mentioned, on jb 7.4 you were able to bypass the issue by selecting the genesis core then back to main menu and “Run cartridge”.

bumfacekilla, 2019-06-13:
I have the same issue using a PAL Mega CD 2, worked fine on JB7.4 aside from needing to enter and exit the Genesis core to get it to kick in.

TheRealFlamepanther, 2019-06-14:
The old behavior was that after reboot, the alternate BIOS wouldn't be loaded until after the Genesis core had been triggered, either by running a cartridge (or CD) or by entering Cores>>Genesis and backing out. The new 7.5 behavior is that the alternate BIOS doesn't load **at all** whatsoever. It's completely broken.

US Sega CD 2.

BarbieOnWeed, 2019-08-10:
> Another user confirmed this issue on the atariage thread on a NA model 2, so it’s not related only to the jpn units.
>
> It’s kinda different because the alternate bios would not run at all, where in [...]

This is what you have to do even on the new 7.6 version.

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks

ChronoStriker1, 2019-11-20:
I can confirm that they do indeed load but due to timing issues all bios's tested crash.