SmokeMonsterPacks/Mega-Sg-Jailbreak #40: Lunar games have video corruption on JB7.5 on Japanese Model 2 Mega-CD

ChronoStriker1, 2019-06-13:
**Describe the bug**
Lunar (USA and Japan) when entering the intro video and the game graphics corrupt.  Lunar 2 (USA, didn't test Japan) graphics for game are corrupt, intro video is OK.  Restoring JV7.4 resolves issue.

**To Reproduce**
Steps to reproduce the behavior:
Load game as normal

**Expected behavior**
Game loads without corruption

**Screenshots**
https://drive.google.com/open?id=1QuSs3f_mvPwppTqdcAgqfia7vEf2rYXf

**Relevant Mega Sg settings**
US BIOS loaded via everdrive due to #39 

**ROM name and hash values**
All Redump (PD) releases - MD5 of zip
Lunar -  The Silver Star (USA) (RE) - f322f76643e47eb91e699d2402a5bb23
Lunar - The Silver Star (Japan) - 87977bc4c5ad0cedf9e036736462f6c3
Lunar - Eternal Blue (USA) - 5e33788c967b2addbdb8792032d12e46

**Additional context**
Confirmed with Kevtris that he did not have the Japanese Model 2 Mega-CD when he created the bugfixes geared towards Japanese Mega-CD's for the official firmware.  I have provided him with pictures of the internal as well as chip model numbers he had requested.

Also tested Un-worked design patched versions of the the above USA version with same results (to be expected)

ChronoStriker1, 2019-08-10:
Confirming issue is still present in JB7.6

ChronoStriker1, 2019-11-20:
Due to CD timing regression in official firmware and JB7.7 issues no CD bios load correctly with my Japanese Mega-CD I can neither confirm nor deny the issue still remains.  Will test after CD timing slider is added to the official firmware.

BooBerry, 2021-02-24:
The CD timing slider was added two days ago in the new firmware update, so you might want to try retesting this (it might work as-is) but if you need to make timing adjustments and you find a working set of timing numbers, please do share them.

ChronoStriker1, 2021-02-24:
I am aware and am getting a new power supply for my mega-sd (needed to use my original for something else more pressing), it should arrive in 3 days.  I will test as soon as I get it.

ChronoStriker1, 2021-03-01:
Unfortunately I am still having the issue where no CD bios will load correctly when the Mega-CD is attached to the Mega-SG.  It is the same behavior on the last two firmware updates.  Japanese bios goes to a grey screen while the music plays in the background.  The cd stops being read almost immediately.  US bios sometimes goes black with the same behavior or it reboots itself every few seconds.  EU bios has severe graphical corruption and then reboots after a bit longer.  I did double check with my us model 2 genesis and it works as its supposed to when connected.

BooBerry, 2021-03-01:
You might want to try adjusting the new CD timing options to see if that fixes it.

ChronoStriker1, 2021-03-01:
I have tried in all possible positions (or at least I think I have) and no change. It would be helpful if there instructions with what each slider did.

GreatHierophant, 2021-06-08:
Lunar : The Silver Star works fine for me in JBv7.8

[image]
[image]

ChronoStriker1, 2021-06-08:
> Lunar : The Silver Star works fine for me in JBv7.8
> 
> [image]
> [image]

Are you running it from a Japanese model 2 Megacd?  And of so with what cd timings and mega cd bios are you using?

GreatHierophant, 2021-06-08:
I was using a US Sega CD Model 2.  I guess the timings between the US and Japanese Model 2s are sufficiently different to allow the game to work in one but not the other.

tjanas, 2023-05-05:
@ChronoStriker1 please re-test with https://www.analogue.co/support/mega-sg/firmware/4.9

ChronoStriker1, 2023-05-07:
Nope still have the same issue

closed 2023-06-26 by tjanas