SmokeMonsterPacks/Mega-Sg-Jailbreak #41: [Master System] [JB 7.5] Bubble Bobble hangs

vanfanel, 2019-06-14:
Hi,

Using "Final Bubble Bobble (Japan)" ROM, the game hangs if left in demo mode.
Just wait for some demo screens, and it will hang.

Any confirmation with cartidge would be welcome.

Brianvgplayer, 2019-06-16:
I had the same issue with the EU and JP ROMs from the smokemonster set, but not with the EU cart. Odd thing is that the game un-hangs when start is pressed (both versions). Hangs usually happened the second or third time level 40 appears in the demo.

vanfanel, 2019-06-25:
I have noticed that the game also hangs while playing: when you get to levels with a lot of sprites on screen, it hangs (and can be un-hanged with the PAUSE button, just like in the demo mode).

BooBerry, 2019-11-19:
This was fixed in the previous v7.6 jailbreak firmware.

vanfanel, 2019-11-19:
Yes, issue can be closed. I will do it. Thanks guys! And specially the firmware fairy XD

closed 2019-11-19 by vanfanel