SmokeMonsterPacks/Mega-Sg-Jailbreak #42: [Master System] [JB 7.5] Rastan Saga: YM music is missing notes.

vanfanel, 2019-06-14:
Hi,

Using "Rastan (USA, Europe).sms", music with YM enabled is missing a lot of notes.
Seems like a timing issue to me.

Any confirmation with original cartidge would be welcome.

Thanks!

SegaSnatcher, 2019-06-15:
Joe from GameSack mentioned SMS FM sounding off in Rastan via Official firmware as well. 

[https://youtu.be/VT2hJE3b_tQ?t=857](url)

Brianvgplayer, 2019-06-16:
I don't have the cart, but the sound in the ROM plays correctly using the Everdrive X7 FM sound instead of the sound from the Mega SG.

vanfanel, 2019-06-18:
> I don't have the cart, but the sound in the ROM plays correctly using the Everdrive X7 FM sound instead of the sound from the Mega SG.

Then the SMS YM implementation is to blame. I do not see how running the game from a cartidge would make a difference, but well, in case someone can confirm...

zxtheproto, 2019-06-20:
> Then the SMS YM implementation on the MegaSG is to blame. I do not see how running the game from a cartidge would make a difference, but well, in case someone can confirm...

The Everdrive X7 has its own emulated synthesizer for YM2413 titles, which it passes through the cartridge audio header of the Genesis. There is also a cartridge adapter with a real YM2413 chip called the Powerbase Mini FM, which behaves similarly.

ultrahealthy, 2019-07-19:
I've had the same problem when using the official firmware, I'm assuming this means the jailbreak doesn't fix it. Another person said the same thing happens with Altered Beast, which also uses FM sound.

In my case, deselecting the Use FM Sound box simply turned the music off altogether, it didn't revert to PSG

skooterblog, 2019-08-19:
Here is my video of Rastan on JB 7.6, from yesterday:
https://www.youtube.com/watch?v=jX9RgFDTF9g

And here is my video Rastan on PowerBase Mini FM (real discrete YM2413), from four years ago, on an original Sega Genesis:
https://www.youtube.com/watch?v=BoAnZH-tsx0

I can't compare them now (workplace, can't hear music), but I hope they can help.

mholzinger, 2019-11-20:
Jailbreak 7.7 - Compared audio from the SMS core to the video referencing GameSack. The FM emulation on the fpga doesn't appear to have any changes, so this might still be an issue.

>> Joe from GameSack mentioned SMS FM sounding off in Rastan via Official firmware as well.
>> https://youtu.be/VT2hJE3b_tQ?t=857

SegaSnatcher, 2019-11-20:
This issue is not simply related to JB.   Kevtris has stated his FM SMS implemtation needs work, but he doesn't have time to fix it anytime soon.

skooterblog, 2019-11-20:
We should NOT buy any Analogue stuff until this bug is fixed. They won't fix if we keep buying their half-baked products.

zxtheproto, 2019-11-20:
I wouldn’t go that far. It’s certainly an issue, but in the grand scale of things it’s pretty minor.
I know it can be obnoxious, but have patience. A fix is in progress.

tjanas, 2023-06-25:
@vanfanel Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

vanfanel, 2023-06-25:
Yes, the same issue happens with JB 7.9 FW. Opened issue there.

tjanas, 2023-06-26:
Moved to https://github.com/analoguejb/Analogue-Mega-Sg-JB/issues/3

closed 2023-06-26 by tjanas