SmokeMonsterPacks/Mega-Sg-Jailbreak #43: Zero Delay audio distortion issues with Test Drive II - The Duel, Onslaught, Populous, & Budokan - The Martial Spirit

zxtheproto, 2019-06-20:
**Describe the bug**
When playing these titles on the Mega SG in the Zero Delay mode, the pitch shifts significantly when playing music. This issue does not exist when using either of the buffered video modes. Populous, Budokan, and Onslaught have only been tested with ROMs, while Test Drive II: The Duel has also been tested on cartridge.

**To Reproduce**
Steps to reproduce the behavior:
1. Toggle setting 'Buffer Mode' to Zero Delay.
2. Load ROM of selected game.
3. Run through title sequence.
4. See (or, more hear) error. You can get a better view of how off-pitch this is by switching back and forth with the Fully Buffered mode.

**Firmware version**
This issue exists with both the most recent official firmware (v4.5) and jailbreak firmware (v7.5). In my testing, previous firmware releases have also exhibited the same issue.

**Relevant Mega Sg settings**

Buffer Mode:

(From the Mega SG Reference Manual v1.3)

_The buffering modes on Mega Sg address the issue that the Genesis/Mega Drive’s native NTSC frame rate is 59.92275fps and the PAL rate is 49.70146fps.  A CRT can display these frame rates with ease but modern displays and HDMI interfaces expect either 60fps. 59.94fps or 50fps and are generally not tolerant of refresh rates that vary from too far from the official frame rates. Mega Sg has three modes that deal with the frame rate differential in different ways.
Fully Buffered Mode buffers full frames to maintain the timing of a 59.92 frame rate using only 60fps.  This mode avoids tearing at the cost of latency. Mega Sg must render at least 1 frame ahead  of the game’s internal rendering to stay ahead.  
Zero Delay Mode speeds Mega Sg up to achieve  a true 60fps frame rate, a speed difference of 0.13%. There is no latency penalty with this method but this method causes Mega Sg to jump ahead by about 1 second every 10 minutes compared to an original Genesis/Mega Drive.  
Single Buffer Mode is something of a compromise between the two methods described above.   Like the Full Buffer option, the original timing is being generated within Mega Sg.  Unlike the Full Buffer, only a portion of the next frame is being prerendered, giving latency of no more than 1 frame depending on when the player activates an input. The drawback is a recurring retrace line that is visible once per every several seconds._

Note that this setting only appears when the system is in Advanced Mode.

**ROM name and hash values**

Test Drive II: The Duel has also been verified with an original cartridge.

File Name: Test Drive II - The Duel (USA, Europe) (Unl).md
MD5: 6995faa82bbb2ff6a985a1505d1092d8
SHA1: cb70e5de149521f20723413cd11c5e661ec63c3e

File Name: Budokan - The Martial Spirit (USA).md
MD5: 4d247a5fc611e66dbe46f960292e3656
SHA1: 93bc8242106bc9b2e0a8a974a3f65b559dd2941d

File Name: Populous (USA).md
MD5: 23495b29851e9d495939a88b627ce2cd
SHA1: 89907c4ba4fd9db4e8ef2271c0253bb0e4b6d52d

Onslaught was tested both with and without the TMSS patch:

File Name: Onslaught (USA, Europe).md
MD5: 455e836ca5dd9cb782e7455e584cf65a
SHA1: dc542ddfa878f2aed3a7fcedc4b0f8d503eb5d70

File Name: Onslaught - TMSS Bypass Patch (Unl) v1 Keeper7.bin
MD5: 068e9a81aa4dbdfaa43999bde4285a26
SHA1: 52befa74d262dc7474720003b860edc1e07a191a

**Additional context**
This is often notable at transitions, which create a sharp pitch bend if music is playing between shots. However, this isn't consistent, and several games (notably Populous and Test Drive II) shift pitch heavily on static screens. 

It should also be stated that this has also been tested not to occur on the original hardware, through referencing screen and audio recordings on the previous games and with personal tests with Test Drive II: The Duel.

frederic-mahe, 2019-06-21:
Very thorough bug report, thanks @zxtheproto !

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks