SmokeMonsterPacks/Mega-Sg-Jailbreak #44: Please add cart dumping to the Mega Sg.

amoore2600, 2019-06-24:
Given all the different cart types the Mega Sg can play, adding cart dumping would be a nice bonus. On the Super NT I have really enjoyed backing up my library and using my original copies between my devices. It's also been wonderful to back back up may save ram too. The only other option I wish existed would be to inject the the save ram back on to the cart as this would make swapping batteries much easier.

WNivek, 2019-09-13:
I concur; the SuperNT jailbreak's cart dumper was a fantastic feature, and one I would love to see translated to the MegaSG jailbreak.

TheRealFlamepanther, 2019-09-19:
Yes please!

frederic-mahe, 2019-09-20:
Duly noted. I am going to close that issue for now.

closed 2019-09-20 by frederic-mahe

justinoue, 2020-06-09:
Is this something that's still in the pipe?