SmokeMonsterPacks/Mega-Sg-Jailbreak #45: Hardball '95 - "SRAM failure"

zxtheproto, 2019-06-25:
**Describe the bug**
When loading Hardball '95 from the Genesis core, it gives an error message when trying to boot the game against a red screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Load ROM 'HardBall '95 (USA).md'.
2. Run game past SEGA logo.
3. See error message - should read as such:

"Cartridge Error Message:
SRAM failure
Press RESET to restart."

**Expected behavior**
The red screen is expected to either not appear or appear very briefly before continuing to the game, without an error message. (See [here](https://www.youtube.com/watch?v=OXwG-xZQ1bs) or [here](https://www.youtube.com/watch?v=_yidJsBiKMc).) The short red screen appearance seems to be more common in emulators, though neither seems to impede gameplay.

**Firmware version**
This was tested with the newest jailbreak firmware as of this writing, JBv7.5.

**ROM name and hash values**

File Name: Hardball '95 (USA).md
MD5: e5b12f2b4e4ff18a838bfec88ec0a601
SHA1: 6a63fba59add9ba8e1845cbfcf4722833893113f

bumfacekilla, 2019-06-25:
Can confirm the same error on JB7.4. Using the ROM identified as 'Hardball '95 (U) [F1]' by GoodTools works normally.

frederic-mahe, 2019-06-26:
OK, this could be a bad ROM issue. The ROM used by @zxtheproto is marked as unverified by [dat-o-matic](https://datomatic.no-intro.org/?page=show_record&s=32&n=0584). A re-dump might be necessary. Until then, maybe someone could try with an original cartridge?

zxtheproto, 2019-06-26:
Just ordered a copy of the cartridge to test; it should be here by July 1. 

I did locate a copy of the SRAM patched ROM, and while it does boot it also doesn't save. Moreover, some elements appear to be corrupt and unalterable, such as the tournament names and logos accessible from the settings.

zxtheproto, 2019-06-30:
The SRAM features do appear to work correctly on an original cartridge:

[image]

[image]

I also tried ripping my own cartridge to a ROM image with my Retrode. 

File Name: HardBall '95 (1995).bin

MD5: e5b12f2b4e4ff18a838bfec88ec0a601
SHA1: 6a63fba59add9ba8e1845cbfcf4722833893113f

This ROM image also doesn't boot on the Mega SG, giving the same error.

I should also note that, while emulators are not a baseline for real hardware performance, they can read and write to the game's SRAM. This, along with the identical hash values between the two ROM images, implies that the original dump is likely correct.

BooBerry, 2019-11-19:
This should be fixed in the new jailbreak firmware released today. :)

closed 2019-11-19 by SmokeMonsterPacks