SmokeMonsterPacks/Mega-Sg-Jailbreak #46: Turbo OutRun (Mega Drive) graphic issues with the sky

BarbieOnWeed, 2019-07-01:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**
The sky and clouds have graphical glitches.

**To Reproduce**
Load the rom and wait for the demo to start. 
You can see the bug on the first and third time the demo is playing.
(It does the same then playing the game)

**Expected behavior**
On one level, lines go straight up from the cloud,
and another, there are lines all over the top screen.

**Screenshots**
https://imgur.com/GhPS8Mu
https://imgur.com/8mJtP12

**Firmware version**
msg_firmware_verJB7.5

**Relevant Mega Sg settings**
Tested different region.

**ROM name and hash values**
Turbo OutRun (Japan, Europe).md
CRC32: 0C661369

frederic-mahe, 2019-07-01:
OK, this is a [valid ROM](https://datomatic.no-intro.org/?page=show_record&s=32&n=1464). Could someone please confirm the graphical glitches? or eventually try with an original cartridge (I am traveling and I don't have access to my Mega Sg).

bumfacekilla, 2019-07-01:
Enabling dither blending in the video menu extra features stops the striping effect at the top of the screen, so possibly not an actual bug.

Tested with original PAL cartridge and does not boot, just black screen. Mega SG is still responding as I can access the menu. The cartridge works perfectly on my region modded PAL Multi-Mega console when set to E, U or J regions. 

One strange thing, this was one of few carts that I was unable to dump using my Retrode, the resulting ROM is always corrupt and unplayable, possibly bad cart but no other to test.

BarbieOnWeed, 2019-07-01:
I was going to make a replie about the graphical bug in the 3rd demo.
But you were faster. Hehe

BooBerry, 2019-11-19:
This should be fixed in the new jailbreak firmware released today. :)

BarbieOnWeed, 2019-11-19:
It is fixed! Just tested it with firmware 7.7. :)

closed 2019-11-19 by SmokeMonsterPacks