SmokeMonsterPacks/Mega-Sg-Jailbreak #47: Auto 50/60Hz video output based on Auto Region Detection.

bumfacekilla, 2019-07-07:
Whilst the automatic region detection feature works great, it can conflict with your saved video settings. 

If playing PAL games at 60Hz output or vice-cersa, there is some visible shaking due to the mismatch. Of course, this is easily fixed by changing to a 50/60Hz option from the Video menu, but must be done each time a PAL/NTSC title is played for smooth output if in conflict with your saved video resolution settings

Would it be possible as an extension to the Auto Region Detection option, to automatically switch video output to 50/60Hz depending on whether a PAL or NTSC title is detected? Perhaps with the ability to pre-select a preferred video mode for each region?

ultrahealthy, 2019-09-21:
I was wondering, have you ever been able to get Master System games to play in PAL (50hz) mode without any graphical shaking or breakup? I'm using the official Firmware 4.5 and there is always graphical breakup at the top of the screen when playing in PAL mode, I've checked with pretty much every western Master System game I own. Is there some setting which only exists in the jailbreak firmware that can fix this?

robc, 2019-11-02:
@ultrahealthy I've been able to - the downside is that you basically need to set the MSg to use a PAL video mode, *then* start your SMS game, then switch it over to PAL.

Switching it to PAL before booting affects the default Mega Drive/Genesis core, and so when you boot an SMS game, it loads its core in, resetting the region to whatever it has saved.

closed 2019-11-20 by frederic-mahe