SmokeMonsterPacks/Mega-Sg-Jailbreak #49: Ultima IV SMS FM Audio Issues

bumfacekilla, 2019-07-12:
Ultima IV for Sega Master System has strange audio glitches when running with 'Use FM' enabled, PSG audio sounds fine. These glitches manifest as buzzing or elongated tones when certain sound effects are triggered and at seemingly random (but frequent) times.

Tested in both PAL mode at 50 Hz output and NTSC Mode at 60 Hz output. Tested with genuine cartridge and with ROM dumped from said cartridge verified in GoodTools. Same ROM was also tested in Genesis Plus GX v1.7.4 core for retroarch with FM Audio enabled for comparison, audio sounded normal. Unfortunately, I do not have any FM audio enabled original Sega hardware to test this with. 

**To Reproduce**
1. Toggle setting 'Use FM' in Core Options.
2. Run ROM (or original cartridge) 'Ultima IV - Quest of the Avatar (EB) [!].sms
3. Noticeable from Start and seems to happen most frequently in battle, it seems as if certain sound effects 'clash' with the music. If left alone, music sounds mostly normal. Seems to be slightly more prevalent when playing in NTSC 60 Hz.

**Expected behavior**
FM audio without glitches

**Firmware version**
Tested on JB7.4 and JB7.5

**Relevant Mega Sg settings**
'Use FM' enabled in SMS Core.

**ROM name and hash values**
Ultima IV - Quest of the Avatar (EB) [!].sms
md5: c54c1a2559cec189f53e794177d002af
sha1: a90e21e5961bcf2e10b715a009c04e7c2017a3b1
crc32: b52d60c8

bumfacekilla, 2019-08-08:
A small update: I have now tested the same ROM on original hardware and Mega SG using a Terra Onion MegaSD which can provide FM audio for supported SMS titles. On original hardware, the FM audio plays perfectly, on Mega SG the issue persists. This may not strictly be a fair test, as the MegaSD is not original FM hardware, but the console is.

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

bumfacekilla, 2019-11-20:
Re-tested on Official 4.7 and JB 7.7. Sadly, the music / sound is still exactly the same as before. Maybe it's the same issue affecting SMS Rastan in issue #42?

Brianvgplayer, 2019-11-21:
The sound also seems to go to PSG with the Mega Everdrive v2 with external cart audio on in the SMS core when a game is started with FM turned off on the Mega SG and turned on in the Mega Everdrive v2 options. Mega Everdrive v2 FM used on work on earlier Mega SG firmware.

tjanas, 2023-05-05:
@bumfacekilla please re-test your cart with https://www.analogue.co/support/mega-sg/firmware/4.9

bumfacekilla, 2023-05-05:
> @bumfacekilla please re-test your cart with https://www.analogue.co/support/mega-sg/firmware/4.9

Re-tested after updating to 4.9, the issue still exists, but is nowhere near as bad. 

The buzzing and odd noises heard before updating still happen, but now much more rarely. The audio is still not perfect, as there are still missing sound effects. For example, the walking sound should be much more rhythmic, but it sounds stilted and uneven like some of the footsteps are not being processed or clashing with other sound channels, almost like a timing issue. 

This is still an improvement over the previous FW's sound nonetheless.

tjanas, 2023-06-25:
@bumfacekilla Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

bumfacekilla, 2023-06-25:
> @bumfacekilla Please re-test with JB7.9 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Mega-Sg-JB

Re-tested after updating to the new JB7.9. The sound in Ultima IV, whilst still not perfect is now a ton better than it was in previous JB revisions. The odd, corrupted noises are now completely gone but the sound is still not quite right. There are still inconsistencies with the volume and timing of most sounds whilst playing, this may sound bad but it really is still a huge improvement over the previous versions.

However (and I thought this very interesting) the FM audio in Wonder Boy III the Dragon's Trap is now broken up and was always fine in previous revisions. It's as if one of the FM channels' data is being read incorrectly, it sounds like a quiet track of random beeps and tones playing throughout any music. What's interesting is that it has messed up in exactly the same way as the FM audio output from the Mega SG cartridge by TerraOnion which plays Ultima IV's FM audio just fine. I wonder if there is some kind of connection where fixing one breaks the other?

tjanas, 2023-06-25:
@bumfacekilla please inquire at the other repository for visibility

closed 2023-06-25 by tjanas