SmokeMonsterPacks/Mega-Sg-Jailbreak #50: Lost Vikings Multitap Detection Issues

clockfort, 2019-07-13:
**Describe the bug**
The Genesis game The Lost Vikings has an option screen that lets you switch the game to one, two, or three players. It has some sort of detection behavior that only lets you select three player mode when there are 3 controllers plugged in via a (Sega, not EA style) multitap.
On Mega SG, it only lets me select one or two players.
The exact same setup of multitaps, controllers, etc with an actual Sega Genesis lets me select 3 player mode.

**To Reproduce**
I can repro this behavior both on everdrive (with the in-game hooks turned off, because weird button combos from the multitap often hit the the right combo) and with an actual cart.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Firmware version**
JB 7.5

bumfacekilla, 2019-07-14:
Enabling passthrough mode under "Hotkeys & Controllers" fixes this. 

Start the game with the Multitap switch set to 'A', enter the menu and enable the passthrough mode. Exit back to the game and slide the Multitap switch to 'Multi' and you're good to go.
Tested and confirmed on JB7.4 & JB7.5

clockfort, 2019-07-14:
Thank you very much!
The official Mega SG manual could probably benefit from mentioning under Passthrough Mode that it might be useful for multitap users; it seems to state that it's only useful for weird master system stuff which is apparently not true.

closed 2019-07-14 by clockfort