SmokeMonsterPacks/Mega-Sg-Jailbreak #51: Zsenilia has graphical bugs. Doesn't display text properly.

SomethinSeemsFishy, 2019-07-19:
**Describe the bug**

The text in Zsenilia is not displaying properly on the MSG. Instead of text, garbage graphics appear.

**To Reproduce**

Steps to reproduce the behavior:
1. Load the Zsenilia ROM
3. As soon as the "RESISTANCE" splash screen fades away, the bug will be visible. 

**Expected behavior**

Zsenilia is a very short demo that displays a logo and a series of balls moving around on top of an animated letterboxed background. Below this, white text should appear on a back background, refreshing at regular intervals to display several short messages. 

The white text does not appear at all on the MSG. Instead, scrambled graphics (they appear to be part of the logo) flash quickly across the screen at regular intervals when the text should be updating.

**Screenshots**

[image]
The garbage graphics can be seen in this image.

**Firmware version**

JB7.5

**ROM name and hash values**

ZSENILIA-BY-RSE-2015.bin (this is the name of the file if you download it directly from the authors)

OR

Zsenilia Intro (PD) Resistance 2015.md (This is what it is called in the SmokeMonster Packs)

The ROM is free and can be found here http://www.pouet.net/prod.php?which=66742

The site also includes links to the source code (https://github.com/ResistanceVault/demo-Zsenilia) and a video of what the demo looks like on real hardware (https://www.youtube.com/watch?v=pu8H4f7P7DU)

**Additional context**

I have only tested this ROM using the Jailbreak (where it displays incorrectly) and BlastEm (where it displays correctly). User the_cavalry on the Classic Gaming discord has tested it on a flash cart and confirmed that it works properly on real hardware (a Genesis model II) but not on the MSG.

frederic-mahe, 2019-11-24:
hi @Squidding could you please confirm that Zsenilia plays correctly when using the latest jailbreak firmware?

Squidding, 2020-01-30:
Apologies for the late response. It plays perfectly now.

closed 2020-01-30 by Squidding