SmokeMonsterPacks/Mega-Sg-Jailbreak #52: Savestates support

exolon, 2019-08-04:
I guess there's a reason the fw doesn't support savestates, but in my mind that's the only thing missing from the firmware and the device itself... Having to use the Sg along with a X7 is overkill, just to have save states support.

Any chance of adding such functionality?

RobotLimeLtd, 2019-09-10:
The X7’s savestate feature doesn’t work well at all with the Mega-SG, resulting in corrupted save data about 50% of the time.  There is a Mega-SD cart that supports Sega CD images, mega drive and master system... and a working savestate feature that’s Mega-SG compatible.

But a firmware based system would be far better, as it could avoid “hotkey clash”.

exolon, 2019-09-11:
Well I have customized the Sg's hotkeys so they don't clash, unless you mean something else?

Also, due to lack of time I actually haven't tried saving on the X7 yet. What you say is very interesting, I should check it out..

But really, it'd be awesome if the fw supported save states - is there a reason not to?

Strongylidium, 2019-09-25:
Firmware based savestates would be a neat implementation. It'd be amazing if feasible for both sd card loaded roms and cartridges. 
Any chance of it appearing in a future update? =)

HoseN16, 2019-09-29:
I'm new to this firmware,(and used to own a X7),so right now there is no save feature for Genesis Roms?

RobotLimeLtd, 2019-09-29:
> Also, due to lack of time I actually haven't tried saving on the X7 yet. What you say is very interesting, I should check it out..

Landstalker (US version) is a game that should reproduce the issue very easily.  It’s on loading the save that you will see the problem.  Background music and sound effects still work, but the main sprite is invisible and the background layers are jumbled up.

RobotLimeLtd, 2019-09-29:
> I'm new to this firmware,(and used to own a X7),so right now there is no save feature for Genesis Roms?

There’s no save feature in the SG firmware, and there are issues saving with the X7.  Not sure if it’s the combination of X7 and SG, or whether the same problem happens with X7 and an OD Megadrive.

HoseN16, 2019-09-29:
Yes there was some random issues with saving on X7 on Mega SG,but I'd say for majority of times save/load state worked fine.
I never had any issues with save/load state using X7 on real hardware (or even clone systemssuch as Mega Retron).
I sold my X7,and now only use Mega SG with this JB firmware.adding Save feature will be really nice and definitely important!

Strongylidium, 2019-09-30:
If firmware savestates are implemented, then the Mega Sg will be light years ahed of traditional emulation as a definitive option for a Sega 8-16 bit experience. The lack of such feature is its only shortcoming imho. Again, I must emphasize that savestates should be implemented having in mind both games loaded from sd card and from actual cartridges. That would be AWESOME!

HoseN16, 2019-09-30:
Load/Save state feature is available on Mega SD which is also an FPGA system.so not having it on Mega SG is only laziness and lack of effort from Analogue IMO.
I wonder if this is something that is possible via this jailbreak firmware independent of Analogue,since Analogue are unlikely to add it.

closed 2019-11-20 by frederic-mahe