SmokeMonsterPacks/Mega-Sg-Jailbreak #53: Gauntlet IV : "Transparent Obstacle" skips in the beginning of the song

MP2E, 2019-08-09:
**Describe the bug**
Play the song "Transparent Obstacle" with the latest official Mega SG firmware, v4.6 as of writing. It should be a smooth "ramp up" sound, but it skips noticeably

**To Reproduce**
Load vgm for gauntlet IV in smokemonsters pack

**Expected behavior**
No skips in audio

**Firmware version**
v4.6 official

**Relevant Mega Sg settings**
Even audio filters seemed to have the issue

**Additional context**
Needs to be tested in Gauntlet IV to make sure it's not a music player bug. However, it sounds fine on my Genesis model 2 or Mister

frederic-mahe, 2019-08-09:
For future reference, this is the tested ROM and its hash values:
```
Gauntlet IV - VGM Player v3.30.md:
  4612f52735bcb084e935e86e6a7ed56e926c57e3f205f611ba272411f9868f1f
  a0ef110cdb0533856cb28d9273a56d71a6594162
  bd5d7389fbd58e6481f9d4250aad40be
  593f2388
```

zxtheproto, 2019-08-14:
I don't notice the bug with the complete set I made from the project2612 VGM files, so my guess would be that this is a error with the v3.30 player in that set.

MP2E, 2019-08-14:
I'll give that a try and report back!

MP2E, 2019-08-25:
My bad on the late response. It does seem like the newer VGM player plays it with no issues. Also weird, there are very minor skips even on real hardware with the v3.30 player i was using. Given those things, I will close this issue. The Mega SG technically differs from real hardware here, but I don't know that it is worth trying to make a buggy music player have an issue in hte exact same way

closed 2019-08-25 by MP2E