SmokeMonsterPacks/Mega-Sg-Jailbreak #54: Landstalker - Inventory Access Sound Glitch

JCnatorU, 2019-08-11:
**Bug Description**
Sometimes when bringing up the inventory menu by hitting the Start button during gameplay, the game hangs on a note.

**To Reproduce**
Steps to reproduce the behavior:
1. Load ROM 'Landstalker (USA).bin' after selecting the Genesis core in the Mega Sg's Cores menu
2. Load any save slot or start a new game
3. Open the inventory by pushing the Start button 
4. Keep trying step #3 until it hangs on a note

**Expected behavior**
The note emitted from accessing the menu should fade in on every single instance

**Video**
https://www.youtube.com/watch?v=tflalUAOJU4

**Firmware version**
JB7.6

**ROM name and hash values**
ROM name: Landstalker (USA).bin
CRC32: FBBB5B97
MD5: 04AE2B65F3A11A7504339E60735BEED8
SHA1: 24345E29427B000B90DF778965DD8834300A9DDE
SHA256: AD9F49CF2D528EE40FAB74F8687ED908036873E54A7DD30C5DC32286C29FC614

NoWayOuy, 2019-08-17:
I have experienced this issue too.

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

JCnatorU, 2019-11-21:
Still experiencing the glitch in JB 7.7, as demonstrated by the following video.

**Video**
https://www.youtube.com/watch?v=T0Cy5GJBOa0

mykie242, 2021-08-10:
I have experienced this with a real cart on actual hardware, so I don't think it's necessarily an issue with the jb.

frederic-mahe, 2021-08-24:
thanks @mykie242 for reporting that the issue is not jb-specific.

closed 2021-08-24 by frederic-mahe