SmokeMonsterPacks/Mega-Sg-Jailbreak #55: Sega Master System BIOS not working with JB7.6

skooterblog, 2019-08-18:
After I installed the firmware JB7.6 on my Mega Sg, it won't work with BIOS files anymore. It will only work without a BIOS file.

Loading a BIOS file and then a ROM will cause Mega Sg to show some glitches and never play the ROM.

This video shows this behavior:
https://www.youtube.com/watch?v=Pr1BQA-WUwQ

Rageypoo, 2019-08-19:
Followed the steps in the video and can confirm working on previous JB firmware 7.5, not working/corruption on JB firmware 7.6.

skooterblog, 2019-08-19:
Thank you for confirming @Rageypoo .

After a hint from Great Hierophant on Discord I confirmed that some BIOSes are actually working, like those with no games included:
[BIOS] Sega Master System (USA, Europe) (v1.3)
[BIOS] Sega Master System (Japan) (v2.1)

The non-working or glitched ones are most of those with games included, like:
[BIOS] Alex Kidd in Miracle World (USA, Europe)
[BIOS] Missile Defense 3-D (USA, Europe) (v4.4)

Notice that those non-working or glitched BIOSes were working just fine on JB7.5, so something got broken in JB7.6.

Also, notice that the JB7.5 manual states that "The system can boot all the official BIOSes except the USA Store Display Unit BIOS at this time."

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks