SmokeMonsterPacks/Mega-Sg-Jailbreak #56: Corrupt video over firmware 4.4 with Sega CD

Surfnet05, 2019-08-22:
This has been confirmed on 2 consoles i have. Bought a second just to confirm and it came in today. (order 30406 and original order 17308)

This is on a newly recapped (in the last month) Sega CD 1 (bios 1.10)

 Came with Firmware 2.9. Tested. FMV video played correctly!  

Immediately upgraded to Smokemonsters firmware   Jailbreak 7.6.... And the problems (bad  FMV video) started. (Road Rash used in pics but it doesn't matter)

OK, Loaded Analouge official 4.6....Still bad FMV

Jumped to oldest firmware on Analogue website...4.2.  FMV Video was good!!!!

Tried official Analouge  firmware 4.5.....Bad FMV again

Tried official 4.4 Firmware...BINGO. There it is good FMV video again

So 4.4 is the highest Firmware i can go without FMV corruption on at least a Sega CD 1. Confirmed on both Analouge consoles.

Issue forwarded to Analouge

SmokeMonsterPacks, 2019-08-22:
Try cleaning the edge connector on your Sega CD with Deoxit.

Surfnet05, 2019-08-22:
That was done.   Keep in mind happening on 2 consoles

Sent from  iPhone

SmokeMonsterPacks, 2019-08-22:
The problem is that hundreds, if not thousands, of us are using Sega CDs on post-v4.4 firmwares without problems. 

You have tested two different Sega CDs?

Surfnet05, 2019-08-22:
I hear ya.  No two sega cd’s here unfortunately.    Got it recapped and cleaned at 

https://dragonshoardgaming.com/ 
Even sent it back to him to doublecheck his work.  
Just saw similar issue already listed herehttps://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/40 
Wonder if same?  Sounds like it.  

Sent from  iPhone

Surfnet05, 2019-08-22:
Also forgot to mention video curroption only happens thru HDMI on Mega Sg. If sega cd plugged into model1 genesis no video corruption thru RF or other connection    Sure not important but interesting.  

Sent from  iPhone

On Thursday, August 22, 2019, 3:23 PM, Brian Joiner <keepingthemiata@yahoo.com> wrote:

I hear ya.  No two sega cd’s here unfortunately.    Got it recapped and cleaned at 

https://dragonshoardgaming.com/ 
Even sent it back to him to doublecheck his work.  
Just saw similar issue already listed herehttps://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/40 
Wonder if same?  Sounds like it.  

Sent from  iPhone

zxtheproto, 2019-08-26:
Are you using the original power supply that came with the Mega Sg? From what I heard, the console can have some graphical glitches when not getting proper amperage.

Surfnet05, 2019-08-26:
Good question.   So was using the trio
http://www.retrogamecave.com/shop/4577155957/trio-m1cd1-power-supply-(g1-cd1-32x)/10506015 
Only using cd power cord off it. 
But I did try 2 sega 1602’s Power supplies.  Same issues 

Sent from  iPhone

zxtheproto, 2019-08-26:
Are you using the original USB block that came with the Mega Sg, as well? Some wall chargers and TV USB ports don’t deliver enough power for the Mega Sg in certain scenarios.

Surfnet05, 2019-08-26:
Yes. Yes I am

Sent from  iPhone

zxtheproto, 2019-08-26:
Do you have a unit currently running JB v4.6? There’s something I’d like you to try.

Surfnet05, 2019-08-26:
Do you mean 7.6?  I have 2 systems.  One running Analogue 4.4 and one running jb 7.6

Sent from  iPhone

Surfnet05, 2019-08-26:
Also to add more info. I have a Mega SD cart. Sega cd games run flawlessly thru it on any firmware

Sent from  iPhone

On Sunday, August 25, 2019, 8:14 PM, Brian Joiner <keepingthemiata@yahoo.com> wrote:

Do you mean 7.6?  I have 2 systems.  One running Analogue 4.4 and one running jb 7.6

Sent from  iPhone

zxtheproto, 2019-08-26:
Get a copy of the Sega CD 1.00 US bios and put it on your SD card in the /BIOS directory. Try adding it with the Alternate Sega CD BIOS loader under Settings -> Core Options, and test with Run Cartridge to see if that fixes the issue.

Surfnet05, 2019-08-26:
Will do.  Will be later tonight tho. Will report back.

Sent from  iPhone

Surfnet05, 2019-08-26:
No change unfortunately. Tried 1.0. 1.10 and for kicks 2.21 on a Sega CD1 .

Surfnet05, 2019-08-26:
[image]
[image]
[image]

frederic-mahe, 2019-08-26:
Hi @Surfnet05, thanks for taking the time to report the issue you've encountered. I am going to edit some of your messages to remove unnecessary previous-message citations and to fix a typo in Analogue's name. I hope you won't mind.

SmokeMonsterPacks, 2019-08-26:
If the games work fine via Mega SD, it's starting to feel more and more like it's something to do with your Sega CD.

Surfnet05, 2019-08-26:
Very possible.  But what changed between firmware 4.4 and above?  Something did. 

Sent from  iPhone

PreZZ79, 2019-09-14:
I had original jb firmware, and everything was fine, played yesterday and it was ok. Updated to firmware 7.6 today, was playing terminator and noticed right away that fmv was garbled and messy, tried other games and same result. Also using sega cd model 1, can i revert the firmware??

zxtheproto, 2019-09-14:
You should be able to revert to an older firmware version by using the same method as you did to install it.

Just to confirm, JB 7.4 & Official 4.4 work, correct? Or, is only the official 4.4 version working without corruption?

PreZZ79, 2019-09-14:
I revert to 7.4 and its fine. I hope they will ne able to fix this i want to play nhl94 cd, it doesnt work in 7.4

Surfnet05, 2019-09-15:
I have sent this issue to Analogue when I was/is having same issue and they have sent it on to the engineers. Hopefully they get it knocked out

Sent from  iPhone

BooBerry, 2019-11-20:
Is this still an issue with the new v7.7 firmware?

Surfnet05, 2019-11-21:
Just tried official 4.7 and Jailbreak 7.7 and the issue is fixed in both versions.  :-)

SmokeMonsterPacks, 2019-11-21:
Thanks @Surfnet05!

closed 2019-11-21 by SmokeMonsterPacks