SmokeMonsterPacks/Mega-Sg-Jailbreak #57: Video noise when playing SMS games

Leonprotos, 2019-09-11:
**Describe the bug**
Mega SG producing video noise on SMS games and some Sega CD games (Though it's way worse on SMS games) I get pixelated dots that flicker around the screen, and I've never had these before. Only happens with firmware above 4.4 and its jailbreak equivalents.

I have already tried different controllers, power, and HDMI cables and the problem persists. Furthermore I have tried using the games on real hardware and these issues do not present themselves.

**To Reproduce**
Steps to reproduce the behavior:
1. Toggle setting 'Issue happens regardless of settings'
2. Load ROM 'Games tested were Zillion and Rocky on SMS'
3. Play through 'Takes several minutes, but let the game run or play and inevitably the noise will appear, usually on red or blue fields.'
4. See error

**Expected behavior**
After several minutes video noise will appear on SMS games, see examples below

**Screenshots**
[image]
[image]
[image]

**Firmware version**
Official firmware 4.6 or JB7.6, issue happens on both.

**Relevant Mega Sg settings**
Issue happens regardless of settings, but these are the values used:
Resolution:  1080P
horizontal pos=10, 
vertical pos=45
H Value: 1462
V Value: 1200 
enable H interpolation
disable V interpolation

All other settings default

**ROM name and hash values**
Zillion (US)
CRC32	5718762C
MD5	D4BF9E7BCF9A48DA53785D2AE7BC4270
SHA-1	3AAB3E95DAC0FA93612DA20CF525DBA4DC4CA6BA

Rocky (World)
CRC32	1BCC7BE3
MD5	582D58D95C9774CD50555366C253A050
SHA-1	8601927CA17419F5D61F757E9371CE533228F7BB

Rageypoo, 2019-09-11:
will verify later tonight

skooterblog, 2019-09-14:
I'm using JB 7.6. I loaded Zillion from the SD Card and let it running for ~2 hours. Then I did the same with Rocky. I couldn't reproduce this issue.

Rageypoo, 2019-09-14:
I too was unable to verify on both 7.5 and 7.6jb with both games. I also asked for the original ROMS used just in case. After discussing further we found that the problem was able to be reproduced by messing with scanlines with version 7.6. My system was not affected by the scanline issues however, and I was unable to reproduce after following instructions.

frederic-mahe, 2019-09-15:
@Rageypoo @skooterblog thank you for following up on that bug report.

I am going to close that issue but feel free to comment if someone finds a sure way to replicate it.

TheRealFlamepanther, 2019-09-29:
The *.6 firmwares introduced this issue for me too. However, it isn't only SMS games that are affected. Genesis games have the same issue, but it's harder to see. The noise is not so obvious as the examples make it look, because each individual miscolored pixel is present for about 1 frame. In action, it looks like color noise you might see in analog video formats. Mostly purple-pink spots in blue areas, sometimes red-pink spots in red areas. Issue goes away if I roll back to an older firmware.

I heard through the grapevine that Kevtris is aware of the issue and has a possible fix already, but I still thought I should pipe up that this is an actual issue that exists.

SegaSnatcher, 2019-10-10:
I feel this issue should be reopened based on this user's post.

https://twitter.com/vpp/status/1182270128995614722

RowanUnderwood, 2019-10-18:
What TV are you connected to?  I have the same issue on a oled55c9pua.

frederic-mahe, 2019-10-18:
@RowanUnderwood you confirm that you've observed these glitches when playing non-SMS games? I am going to modify the title of the issue to reflect that.

RowanUnderwood, 2019-10-20:
Yes, all of the Genesis (Megadrive) games I tested also had the issue.

Leonprotos, 2019-10-21:
> What TV are you connected to? I have the same issue on a oled55c9pua.

I'm using an LG OLED55C6P

TheRealFlamepanther, 2019-10-25:
> What TV are you connected to? I have the same issue on a oled55c9pua.

Vizio E420d-A0

thomas83lin, 2019-10-28:
I actually just started getting this issue.

I had just updated from 7.5jb to 7.6jb and switched to a different tv. So I'm not sure if this problem was present on 7.5jb or if the tv is causing it. 

I started noticing this while using the Mega SG but once trying to isolate the issue I noticed I'm still getting the flickering pixels even while booting straight to the Mega SG menu selection so I getting them everywhere even with no cart connected or before playing a game.

TV model Samsung UN32J5003AF
update:
Figured I'd do minimum testing before posting 
Switching back TV's didn't fix it.
Downgrading back to 7.5jb seemed to fix the problem!!

SmokeMonsterPacks, 2019-11-19:
This should be fixed in the latest JB update. I will re-open this if not.

closed 2019-11-19 by SmokeMonsterPacks

mholzinger, 2019-11-19:
Monitor: Samsung S27D590
Jailbreak: msg_firmware_verJB7.7.bin

Unable to replicate video noise issues with roms on SD Card or MegaDrive XY