SmokeMonsterPacks/Mega-Sg-Jailbreak #58: MSG JB Filesystem corrupt

Nawfal3D, 2019-09-20:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**

Can not unzip MSG JB Filesystem as "End-of-central-directory signature not found"  

**To Reproduce**
Steps to reproduce the behavior:
1. Download "MSG JB Filesystem"
2. Extract

frederic-mahe, 2019-09-20:
I've checked the zip, there's no error:
```sh
unzip -t MSG\ JB\ Filesystem.zip 
Archive:  MSG JB Filesystem.zip
    testing: MSG JB Filesystem/       OK
    testing: MSG JB Filesystem/SD Card/   OK
    testing: MSG JB Filesystem/SD Card/BIOS/   OK
    testing: MSG JB Filesystem/SD Card/BIOS/BIOS.txt   OK
    testing: MSG JB Filesystem/SD Card/COLECO/   OK
    testing: MSG JB Filesystem/SD Card/GENESIS/   OK
    testing: MSG JB Filesystem/SD Card/GG/   OK
    testing: MSG JB Filesystem/SD Card/readme.txt   OK
    testing: MSG JB Filesystem/SD Card/SMS/   OK
No errors detected in compressed data of MSG JB Filesystem.zip.

zip -F MSG\ JB\ Filesystem.zip --out MSG\ JB\ Filesystemfix.zip
Fix archive (-F) - assume mostly intact archive
Zip entry offsets do not need adjusting
 copying: MSG JB Filesystem/
 copying: MSG JB Filesystem/SD Card/
 copying: MSG JB Filesystem/SD Card/BIOS/
 copying: MSG JB Filesystem/SD Card/BIOS/BIOS.txt
 copying: MSG JB Filesystem/SD Card/COLECO/
 copying: MSG JB Filesystem/SD Card/GENESIS/
 copying: MSG JB Filesystem/SD Card/GG/
 copying: MSG JB Filesystem/SD Card/readme.txt
 copying: MSG JB Filesystem/SD Card/SMS/
```
could you please try that direct link:

https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/raw/master/firmware/MSG%20JB%20Filesystem.zip

Nawfal3D, 2019-09-28:
unzip -t MSG\ JB\ Filesystem.zip 
Archive:  MSG JB Filesystem.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of MSG JB Filesystem.zip or
        MSG JB Filesystem.zip.zip, and cannot find MSG JB Filesystem.zip.ZIP, period.

However, I can extract the direct link!

frederic-mahe, 2019-09-28:
Thanks @Nawfal3D I've fixed the link to the zipped MSG JB Filesystem in the README.

closed 2019-09-28 by frederic-mahe