SmokeMonsterPacks/Mega-Sg-Jailbreak #59: Flickering pixels in Firmware, JB7.6, reverting to 7.5 corrected issue.

RowanUnderwood, 2019-10-15:
(do not submit bugs if you are using any means of powering your Mega Sg other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

**Describe the bug**
Flickering pixels in all tested games in Firmware, JB7.6, reverting to 7.5 corrected issue.  Tried both from cart and from SD card, all other settings stock.  https://i.imgur.com/A7jsz5s.mp4 is what it looks like (this is slowed down for better visibility).  My display is an oled55c9pua, running straight through with provided HDMI cable.
 

**To Reproduce**
Steps to reproduce the behavior:
1. install firmware 7.6
2. Load any rom / cart
3. pixels flicker

**Expected behavior**
Nice sharp pixels like in 7.5 :)

**Screenshots**
https://i.imgur.com/A7jsz5s.mp4

**Firmware version**
Only submit reports for the latest jailbreak or official firmware (please be ready to retest your issue if the firmware is updated).

**Relevant Mega Sg settings**

**ROM name and hash values**
On Mac and Linux, use command line tools to get the hash values of your ROM (for example: `sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility. If you can, please indicate if the bug occurs when using an original cartridge or CD as well.

**Additional context**
Add any other context about the problem here:

frederic-mahe, 2019-10-17:
@RowanUnderwood I took the liberty to trim your post. Do you get these flickering pixels only in SMS games or in Megadrive games too? Is your issue related to issue #57 ?

RowanUnderwood, 2019-10-18:
Yup I believe it's the same as #57, sorry about that, didn't see that one
in there.  Didn't get a chance to test any SMS games on this firmware
revision, but saw the noise in every Mega Drive game I tried, including
real carts and from SD card.  We should ask Leonprotos what TV he is using,
maybe it's related to that?  Mine is a oled55c9pua running in game mode.

frederic-mahe, 2019-10-18:
No problem. I am going to close that issue.

closed 2019-10-18 by frederic-mahe