SmokeMonsterPacks/Mega-Sg-Jailbreak #60: Cheat Code Files

4ch1m, 2019-10-15:
Hi there,

I'd like to bring this topic back up again: [Save cheat codes and button option](https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/14)

I understand that the MegaSG is meant for purists and enthusiasts; trying to recreate the original experience as faithfully as possible.
Cheat codes may not have been a standard feature back in the days.
However, we nowadays simply don't have the time to practice a single game for weeks any more.
So, the use of cheat codes is the only way to enjoy these old games without wasting too much time.
(Aside from save-states; which obviously are out of discussion due to the fact that the MegaSG is not based on software-emulation.)

Therefore I also think that it should be possible to permanently save cheat codes for individual games; not having to re-enter the codes with every play-session.

For example:
Both the RetroN5 and the RetroFreak offer "cheat databases" (based on ROM checksums) for ease of use.

see:
- [http://retron5.in/node/6](http://retron5.in/node/6) ([RetronCheats-v1.2.zip](http://www.retron5.in/downloads/RetronCheats-v1.2.zip))
- [https://www.cybergadget.co.jp/support/retrofreak/codefreak/](https://www.cybergadget.co.jp/support/retrofreak/codefreak/) ([retrofreak_code.zip](https://www.cybergadget.co.jp/assets/files/download/retrofreak_code.zip))

I totally understand that it would be an extra effort to implement something like this for the MegaSG.
But please give us at least a simple way to store cheat-codes on the SD-card.

Suggestion:
Let's put CHT-files alongside the actual ROM-files.
These CHT-files simply contain six lines holding the cheat-codes; which then can be used to automatically populate the cheat-menu when loading a ROM.

e.g.
```
'Mickey Mouse - Castle of Illusion (U) [!].smd'
'Mickey Mouse - Castle of Illusion (U) [!].cht'
'Thunder Force 3 (JU) [!].bin'
'Thunder Force 3 (JU) [!].cht'
...
```

Should be fairly simple to implement.

Unfortunately, the MegaSG-firmware/-system is not open-source compared to projects like MiSTer.
I'd love to contribute; but I guess we're stuck with this feature-request for now. :)

Thanks!

exolon, 2019-10-16:
Oh I agree, that would be an excellent addition, no matter how it's implemented!

closed 2019-11-20 by frederic-mahe

frederic-mahe, 2019-11-20:
@4ch1m Our policy is to keep feature request open for a while, but not indefinitely, as the goal is to focus on actual issues. Your suggestion is still available when searching with the feature request tag, closing doesn't mean deleting. Thanks again for taking the time to write a detailed description of a potential feature.